kliment/Printrun #1263: using ;@exit in gcode does not exit

fequalsf, 2022-05-18:
I am using python in another software to open pronsole (using subprocess.Popen) to run gcode and I want to exit the pronsole window when it is finished sending gcode.

Using ;@exit, when pronsole reaches the ;@exit command, it asks in the console if I really want to exit....

G-Code calling host command "exit"
Disconnecting from printer...
Are you sure you want to exit while printing?
(this will terminate the print).

How can I exit the console from gcode or cli arguments, without user input?

Thanks!

volconst, 2022-05-19:
Doesn't pronsole finish if you close the file handle?

fequalsf, 2022-05-20:
Hello, 

Im not sure if i understand the question...

I am running pronsole from another software suing these arguments:

C:\Printrun-Win-Slic3r-03Feb2015\Printrun-Win-Slic3r-03Feb2015\pronsole.exe
-e
connect
-e
block_until_online
-e
load C:\Printrun-Win-Slic3r-03Feb2015\Printrun-Win-Slic3r-03Feb2015\print.gcode
-e
print

Then, I want pronsole to automatically close the console when it is finished. So I used ;@exit as the last line of the gcode in the file, but what I get is this response asking for user feedback:

G-Code calling host command "exit"
Disconnecting from printer...
Are you sure you want to exit while printing?
(this will terminate the print).

So, obviously the answer is yes, I want to exit, but is it possible to do this without user feedback?

What I understand is, once it gets to the last line of code, the machine has already sent "ok" so all the gcode is already in the machine buffer.

So at that point it should be safe to exit the console without any user input?

Sorry if this still doesnt make sense, and thank you for the help! And no, I'm not a programmer, so I'm hoping for a solution that is as simple and stupid as me.

Cheers!

volconst, 2022-05-21:
Do not send ;@exit and use this command

```
C:\Printrun-Win-Slic3r-03Feb2015\Printrun-Win-Slic3r-03Feb2015\pronsole.exe
-e connect
-e block_until_online
-e load C:\Printrun-Win-Slic3r-03Feb2015\Printrun-Win-Slic3r-03Feb2015\print.gcode
-e print
-e "shell if self.p.print_thread:self.p.print_thread.join()" -e exit
```
The pronsole command 'exit' is added, which is almost the same as the gcode host command, except that it would execute immediately after the print is started. The 'thread.join' thing makes it wait for the end of the print

fequalsf, 2022-05-21:
This is closer... still does not seem to work.

Does this line need to be in quotations like you have written it?

"shell if self.p.print_thread:self.p.print_thread.join()"

I'm attaching an image of the sequence of arguments sent to pronsole

And an image of the console. It still ends with a "y/n" and, of course, typing y + enter  will close the console. But it will still not happen automatically.

Thank you for the help!

[image]
[image]

fequalsf, 2022-05-21:
SORRY!!!

I still had ;@exit at the end of my gcode.

it is working perfect now!!!!

many many thanks!!!!

volconst, 2022-05-22:
Great :)
What IDE is this?

fequalsf, 2022-05-22:
This is using grasshopper, a visual programming environment for rhino. https://www.grasshopper3d.com/

I will share the code I've made here for anyone else who might find this

fequalsf, 2022-05-26:
Here is the code I use in Grasshopper, in case anyone finds this later. Maybe not the cleanest, but I'm not a programmer.

I will make a component later that will be shared on food4rhino.com:

import subprocess, shlex
import Grasshopper as gh
import os

dirPront = dir + "pronsole.exe" #pronsole directory

dirGcode = dir + "print.gcode" #write file path

def fix_load(commands):
    index = arguments.index("load")
    load = " ".join(arguments[index:index+2])
    
    del arguments[index:index+2]
    arguments.insert(index, load)
    
    return arguments
def fix_shell(commands):
    index = arguments.index("shell")
    shell = " ".join(arguments[index:index+3])
    
    del arguments[index:index+3]
    arguments.insert(index, shell)
    
    return arguments

# ------------

arg1 = "-e connect -e block_until_online -e load " + dirGcode + " -e print -e shell if self.p.print_thread:self.p.print_thread.join()" + " -e exit"  

commands = dirPront + " " + arg1
arguments = shlex.split(commands, posix=False)

if "load" in arguments:
    arguments = fix_load(arguments)
a = arguments

if "shell" in arguments:
    arguments = fix_shell(arguments)
a = arguments

if Run:
    gPath = open(dirGcode, "w") # Open the file
    for line in gcode: # Iterate through lines
        gPath.write(line + "\n") # Write separate lines
    gPath.close() # Close the file
    subprocess.Popen(arguments, shell=False)

closed 2022-05-26 by fequalsf