kliment/Printrun #1265: Macro Gcode limitation

Br000gle, 2022-05-29:
I'm not sure if this is a error with the application, or just my printers memory limitations because I can't find any documentation stating otherwise, but there seems to be a limit on how many commands you can write within a single macro. Its not limiting how many lines of code I put, but rather the amount of characters I write, or the amount of bytes the specific code takes up. Long scripts are basically aborting mid way for me. I did a test to just send slightly different text to my printer screen a couple dozen times and even that aborts half way

volconst, 2022-05-30:
Are the scripts long, or they contain some complex syntax?
Post a macro which aborts?
What do you mean by abort? Does the printer, disconnect, or an error message is shown?
Enable Debug communication in the menu and track if all the commands are sent and and followed by RECV: ok.

KevWal, 2024-06-24:
I have this same issue.  Pronterface 2.1.0 

A macro like this:

> ! self.p.send("G90") # Set all axes to absolute
> ! self.p.send("G28") # Home all axis
> ! self.p.send("G30 C1 E1 X15 Y15") # Front Left
> ! self.p.send("G30 C1 E1 X20 Y20") # Front Left
> ! self.p.send("G30 C1 E1 X165 Y15") # Front Right
> ! self.p.send("G30 C1 E1 X160 Y20") # Front Right
> ! self.p.send("G30 C1 E1 X165 Y165") # Back Right
> ! self.p.send("G30 C1 E1 X160 Y160") # Back Right
> ! self.p.send("G30 C1 E1 X15 Y165") # Back Left
> ! self.p.send("G30 C1 E1 X20 Y160") # Back Left
> ! self.p.send("G30 C1 E1 X90 Y90") # Center

Stops midway through the "G30 C1 E1 X15 Y165" command:

> Macro 'CheckLevel' defined
> Bed X: 15.00 Y: 15.00 Z: -0.20
> Bed X: 20.00 Y: 20.00 Z: -0.11
> Bed X: 165.00 Y: 15.00 Z: -0.18
> Bed X: 160.00 Y: 20.00 Z: -0.18
> Bed X: 165.00 Y: 165.00 Z: -0.32
> Bed X: 160.00 Y: 160.00 Z: -0.30
> Bed X: 15.00 Y: 160.00 Z: -0.19

I tried this without all the ! self.p.send(" python but get the same result.

Thanks
Kevin

DivingDuck, 2024-06-25:
Can you please attach the log file of Pronterface? What did the log say?

volconst, 2024-06-25:
Maybe the printer buffer gets overflowed and the excess commands are ignored.
Try putting sleep() between the commands
```
!import time
!time.sleep(1)
```

KevWal, 2024-06-25:
I did think that it might be printer buffer related, but why does normal printing respect flow control and the Macro system not?

I tried a few time.sleep(60), but it seems to mess things up with no output being printed till the script has finished, but the script does get to the end.

Cheers
Kev

DivingDuck, 2024-06-25:
60 is quite a lot. Remember my question, what does the log tell you?

volconst, 2024-06-25:
> I did think that it might be printer buffer related, but why does normal printing respect flow control and the Macro system not?
> 
> I tried a few time.sleep(60), but it seems to mess things up with no output being printed till the script has finished, but the script does get to the end.
> 
> Cheers Kev

Maybe the macros do not use the queue and are immediate commands that execute before the queue - one  reason for this is that the user actions may be due to emergency.

volconst, 2024-06-25:
Somewhere in the api may be a method that sends and waits confirmation for the command, Using it instead of self.p.send() may fix the problem.

KevWal, 2024-06-25:
> 60 is quite a lot. Remember my question, what does the log tell you?

It needs to be pretty high, each set of 5 probes takes about 45 seconds, and a possible homing before the first one.

Sorry I forgot the log, but checking it shows nothing more than on screen console display?

Thanks
Kevin

KevWal, 2024-06-25:
I wonder if self.mainqueue.append will help me here, it is what is used to queue a command when the printer is printing:

https://github.com/kliment/Printrun/blob/0af3755bfd543ee6f911ea489f448d8638fd61b9/printrun/printcore.py#L538

So self.mainqueue.append gives a PronterWindow object has no object 'mainqueue' error, and a self.p.mainqueue.append is accepted, but does nothing - the printer doesnt move.

Cheers
Kev

DivingDuck, 2024-06-25:
You can try to enable debug communications in addition. Maybe it give you some additional info from the printer.

Btw, what kind of printer and firmware are you using?