kliment/Printrun #1266: m500 command stuck on sending

Yurtle13x, 2022-05-30:
The command wont go anywhere and its been stuck on the same command for an hour

DivingDuck, 2022-05-30:
Works fine for me

Yurtle13x, 2022-05-30:
hey do you have a discord i can talk to you on because for me the command just wont go to my printer

DivingDuck, 2022-05-31:
No discord here.

There is missing quit a lot of information here. You really should give those information first...

Version of Pronterface (Help)? 
Your operating system?
Commercial or DIY printer 
Your printer firmware (Marlin, Smoothieware, Reprap etc. version)?
How is your printer connected to Pronterface (USB etc.)?
Have you establish a connection between Pronterface and your printer?
--> Can you see a connection report in your console window
--> Are you able to e.g. move the print head with the control buttons
One or two screenshots attached are telling often more then 1000 words.

Something like this:
 
[image]

[image]

[image]

The last picture is telling a lot of requested/needed information:
Successfully connected with printer via USB, printers firmware is communicating with Pronterface and I'm able to communicate with the printer

DivingDuck, 2022-06-11:
@Yurtle13x, any news?

DivingDuck, 2022-06-18:
@Yurtle13x , I will close the issue as there is no further information available from you. Feel free to open the issue again if you need additional help.

closed 2022-06-18 by DivingDuck