kliment/Printrun #1268: Fatal Error Detected: Failed to execute script pronterface

CharlesGibbsNam, 2022-06-11:
Hi, 

This error has occurred under win11 for me.
I was running Win10 with no issues, I think this has to do with the OS structure maybe?

DivingDuck, 2022-06-11:
What version of pronterface and what exactly is the error message?

CharlesGibbsNam, 2022-06-11:
[image]
Hi, I have attached a screenshot, This is the latest version downloaded about an hour ago.
Regards
Charles

DivingDuck, 2022-06-11:
can you pls test this version: https://github.com/DivingDuck/Printrun/suites/6557963620/artifacts/245600823

I guess this is the wxPython issue.

CharlesGibbsNam, 2022-06-11:
HI,

I am downloading now, Will revert shortly.

CharlesGibbsNam, 2022-06-11:
Good news! 

Downloaded and tested, working.

Thank you kindly for your help.

Regards 
Charles

closed 2022-06-11 by CharlesGibbsNam

DivingDuck, 2022-06-11:
@CharlesGibbsNam, you are welcome.

@kliment, I've add a new label wxPython issue as this is the only problem pronterface have for some Windows installations. The shared version above is compiled with one of the latest wxPython snapshot versions.