kliment/Printrun #1274: windows 8.3 ? seriously ?

meutzitzu, 2022-07-08:
Howcome an 8-bit machine is able to handle files with more than 8 characters but a python script running on the host machine can not ? 

How does this make any sense ? 

It is extremely useful to have all the information your slicer provides in the name of the gcode file. limiting yourself to 8 characters is just silly, if not absurd

what shocks me the most is that I've searched through the previous issues and noone seems to have complained about this before 

**???!??!??!????**

lrdiceman, 2022-07-08:
@meutzitzu , if you have recovered from the shock, you may try explain what exactly is the problem, steps to reproduce.
I do not have pronterface running, but I would also be shocked if it cannot open long filenames. Do you claim this?
What operating system and pronterface version are you using?

volconst, 2022-07-09:
I do not see in the code actual checking whether the format is 8.3. Only the message when the number or filenames is not 2 says it must be 8.3.
@meutzitzu , can you test what happens if you enter longer file name with and without spaces?
Did you get this message when you tried to upload filename with spaces?
Please post the command you used

meutzitzu, 2022-07-11:
if I try to upload a longer file I must specify a target (filename on the sdcard) that is in 8.3 format
this works
if I try to upload a file and I specify a target name longer than 8 characters it doesn't upload the file, it makes the printer start printing and says `open failed`
like so:
```bash
ttyACM0 193°> upload /data/work/3DP/Models/sierpinski_pyramid_tester.gcode sierpinski.g
Loading file: /data/work/3DP/Models/sierpinski_pyramid_tester.gcode
Loaded /data/work/3DP/Models/sierpinski_pyramid_tester.gcode, 4742 lines.
Estimated duration: 30 layers, 0:03:39
Uploading as sierpinski.g
Uploading /data/work/3DP/Models/sierpinski_pyramid_tester.gcode
Press Ctrl-C to interrupt upload.
Progress: 00.0%Print started at: 18:42:35
Now fresh file: sierpinski.g
open failed, File: sierpinski.g.
ttyACM0 204° 000.3%^C...interrupted!
A partial file named sierpinski.g may have been written to the sd card.
```
which I have to interrupt because I don't want it to start printing

volconst, 2022-07-11:
Pronterface sends the long filename, so we can say that it supports long filenames.
But the printer does not create the file and pronterface could check this condition and abort the sending.
Can you enable Debug communications and upload again - I need the line(s) containing the error. Please tell your printer firmware model to collect firmware variations of error messages.
The other route is to forbid long filenames :)