kliment/Printrun #1275: Python error.. FYI

brown-richard, 2022-07-13:
The print goes:
- from 0.00 mm to 76.79 mm in X and is 76.79 mm wide
- from 0.00 mm to 66.35 mm in Y and is 66.35 mm deep
- from 0.00 mm to 15.00 mm in Z and is 15.00 mm high
Estimated duration: 108 layers, 0:50:59
[ERROR] OpenGL failed, disabling it:
Traceback (most recent call last):
  File "printrun\gl\panel.pyc", line 116, in processPaintEvent
  File "printrun\gl\panel.pyc", line 231, in OnDraw
  File "printrun\gcview.pyc", line 154, in draw_objects
  File "printrun\gl\libtatlin\actors.pyc", line 713, in display
  File "printrun\gl\libtatlin\actors.pyc", line 749, in _display_travels
  File "pyglet\gl\lib.pyc", line 105, in errcheck
GLException: out of memory

Does this mean anything to someone?

DivingDuck, 2022-07-20:
@brown-richard,
What version of Pronterface are you running and what is your operating system?

DivingDuck, 2022-10-09:
@brown-richard, there was passing quite some time now. I will close this issue for now due to lack of information/response. Feel free to come back if you have further information's so that we are able to help.
Best regards,
DD

closed 2022-10-09 by DivingDuck