kliment/Printrun #1277: TypeError: Pen(): arguments did not match any overloaded call

GkqqNkKC, 2022-07-22:
Hello, I have recently upgraded my Ubuntu from 20 to 22, and now I can't get Pronterface to work at all. It says:

```
WARNING:root:Could not setup DBus for sleep inhibition: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.ScreenSaver was not provided by any .service files
Traceback (most recent call last):
  File "/usr/bin/pronterface", line 62, in <module>
    app = PronterApp(False)
  File "/usr/lib/python3/dist-packages/printrun/pronterface.py", line 2439, in __init__
    self.mainwindow = PronterWindow(self)
  File "/usr/lib/python3/dist-packages/printrun/pronterface.py", line 207, in __init__
    self.reload_ui()
  File "/usr/lib/python3/dist-packages/printrun/pronterface.py", line 281, in reload_ui
    self.createGui(self.settings.uimode == _("Compact"),
  File "/usr/lib/python3/dist-packages/printrun/gui/__init__.py", line 259, in createGui
    viz_pane = VizPane(self, vizpanel)
  File "/usr/lib/python3/dist-packages/printrun/gui/viz.py", line 115, in __init__
    root.gwindow = gviz.GvizWindow(build_dimensions = root.build_dimensions_list,
  File "/usr/lib/python3/dist-packages/printrun/gviz.py", line 77, in __init__
    self.p = Gviz(panel, size = size, build_dimensions = build_dimensions, grid = grid, extrusion_width = extrusion_width, bgcolor = bgcolor, realparent = self)
  File "/usr/lib/python3/dist-packages/printrun/gviz.py", line 197, in __init__
    self.mainpen = wx.Pen(wx.Colour(0, 0, 0), penwidth)
TypeError: Pen(): arguments did not match any overloaded call:
  overload 1: too many arguments
  overload 2: argument 2 has unexpected type 'float'
  overload 3: argument 1 has unexpected type 'Colour'
```

I also downgraded my Python version to 3.6, 3.7 or 3.9, and it still doesn't work. The app is stuck and I have to press Ctrl+C to exit. I also have reinstalled Pronterface. Again, no success

hroncok, 2022-07-25:
This should have been fixed by https://github.com/kliment/Printrun/pull/1224

rockstorm101, 2022-07-25:
Ah, of course, that's it. Thanks a lot @hroncok. I'm afraid the Printrun version on Ubuntu 22.04 is not yet patched for this error. Patch is already in place in testing so it'll make its way trough for 22.10.

Using Python 3.9 as a work around should work though. Remember to change the shebang line on `/usr/bin/pronterface` like:
```diff
- #!/usr/bin/python3
+ #!/usr/bin/python3.9
```

rockstorm101, 2022-07-26:
Duplicate of #1225. Closing.

closed 2022-07-26 by rockstorm101

galagithub, 2023-04-26:
almost 1 year passed an the issue is still present on Ubuntu 22.04 which is an LTS version !!!!

zhennik26, 2023-06-07:
I have the same problem on Ubuntu

DivingDuck, 2023-06-07:
@zhennik26,
as already stated, please contact the Ubuntu team. We can't do anything on this as we do not build the applications for the different Linux distros. 
Alternatively you can run it from source following the instructions: https://github.com/kliment/Printrun#running-from-source.