kliment/Printrun #1278: Bed check during warm up?

Nitrowing, 2022-07-29:
Hi, I've no issues with the program - it's great!
I was just starting a print when I thought, "Why doesn't the software do the Home and bed check while the bed/nozzle is heating up?"
There may be a perfectly obvious reason for this, I have no idea how the instructions are implemented!
I know this wouldn't save a whole heap of time but...

DivingDuck, 2022-07-30:
This isn't the best idea but if you want to do it you can do it. 

You will face out less and unpredictable bed adhesion as long as the bed is not settled to the desired bed temperature. This have to do with thermal material expansion and uneven heat spreading (lead to little and mostly uneven deformations) during the heating phase. The second problem is that the nozzle tip will usually stick some plastic that will give you wrong z values as long it is not melted to squish away while probing.

There are different g-codes available for doing it. Those will be set up usually in your preferred slicer. Look for M140/M190 and M104/M109  (https://reprap.org/wiki/G-code)

Alternatively you can do it manually by typing the g-codes commands in the terminal window or set up custom buttons (https://github.com/kliment/Printrun#using-macros-and-custom-buttons)

Nitrowing, 2022-07-30:
Thank you for the quick and detailed reply.
As I usually print PETG on my i3 Mk2, the nozzle gets up to temp very quickly in relation to the bed (230C & 85C). I really can't see the bed expansion differing between 75C and 85C and the nozzle is already stable before the bed reaches 75C.
I'll have a look at that G-code page...

DivingDuck, 2022-07-30:
Sure it does. For example setup print bed first with M140 S85, then start heater with M104 S240, then wait 15 sec. with G4 S15, then make your homing with G28 and then do what ever you want in addition like bed leveling. After this the last commands should be set temperature for bed and wait with M190 S85 followed by M109 S240 for the extruder.

M140 S85; start heating bed
M104 S240; start heating extruder
G4 S15; wait 15 sec.
G28; homing all axes

... ; do what you want like bed leveling 
... ; do what you want
... ; do what you want

M190 S85; set bed temp and wait
M109 S240; set extruder temp and wait
..,; do what you want like purging line, reset etc.
... ; do what you want
...; and now your are ready to print

Nitrowing, 2022-07-31:
Ah! Very good! Thank you

closed 2022-07-31 by Nitrowing