kliment/Printrun #1279: Error encoding

Humery, 2022-07-31:
In pronterface, when I click on Connect, I obtain this error :
  File "encodings\utf_8.pyc", line 16, in decode
UnicodeDecodeError: 'utf8' codec can't decode byte 0xe8 in position 39: invalid continuation byte

Thank you for your help.

DivingDuck, 2022-07-31:
What Pronterface version and OS are you running?

Humery, 2022-07-31:
I use Printrun 2014.08.01
My OS is Windows 11, Version 22H2

DivingDuck, 2022-07-31:
This version is retired and not compatible with Windows 10/11. 

It seems like the config file is corrupt. Sometimes it helps to delete them. Close Pronterface, make a backup of this file and delete it. With starting Pronterface the config file will be created new. For windows you will have two places for the config file. One is in the root of your user directory like "c:\users\YOUR_USER_NAME" and the second is at "c:\Users\YOUR_USER_NAME\AppData\Local\Printrun". Delete the the file printrunconf.ini as well.

You should better install the latest release 2.0.0rc8 (https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc8) as there are compatibility issues with Windows 10/11. And do not switch between 1.x and 2.x versions as they use the same config files what will lead in an corrupt config file.

Humery, 2022-07-31:
J'ai chargé la 2.0.0rc8.
Maintenant, le message d'erreur concerne le port, qui est pourtant dispo (et que je vois dans "Périphériques" et qui n'est utilisé par aucun autre logiciel.
Connecting...
Could not connect to COM3 at baudrate 115200:
Serial error: could not open port 'COM3': PermissionError(13, 'Accès refusé.', None, 5)

DivingDuck, 2022-07-31:
Guess, this isn't the original error. have you seen an error with wxPython involved? If yes can you test this version for me:

https://github.com/DivingDuck/Printrun/actions/runs/2525453711

It is an interim version that deals with the wxPython problem. For all changes pls. see: 
https://github.com/DivingDuck/Printrun/wiki/My-latest-development-version-build-with-github-workflow

Humery, 2022-08-01:
Now I use version 2.0.0rc8, and I have connected on an other port.
All is right :-) 

Thank you

DivingDuck, 2022-08-01:
You are welcome. I will close the issue.

closed 2022-08-01 by DivingDuck