kliment/Printrun #1280: Runtime Error when turning on Temperature Gauges in Settings

Dardwizzle, 2022-08-01:
Hello,

I get the following Runtime Error when I turn on the Temperature Gauges in Settings / Options. I do notice that the linear gauges do show under the "Print Flow". This leads me to believe that this may be just a warning or informational error rather then an actual failure in the routine as there doesn't appear to be anything wrong with the temperature display on either the graph or gauges.   Nothing else seems to be erroring, and everything does appear to be working with no problem.

Connecting...
Printer is now online.
Traceback (most recent call last):
  File "wx\core.py", line 3383, in <lambda>
  File "printrun\gl\panel.py", line 139, in Refresh
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted

Just downloaded the [Printrun 2.0.0rc8] version

I am running Windows 11 64 Bit. I downloaded the [pronterface-windows-x64_3.8-2.0.0rc8.zip](https://github.com/kliment/Printrun/releases/download/printrun-2.0.0rc8/pronterface-windows-x64_3.8-2.0.0rc8.zip)

If you would like me to provide any additional information, I will gladly do so, if you are already aware of this, my apologies as I did check the issues here before I opened this.  Being new to using this application, I am not to sure what you guys might need to look into this further if anything.

Many thanks and keep up the awesome work!

Cheers,

JD

volconst, 2022-08-03:
I think these messages appear only when the settings are edited and are harmless.
They could be prevented by checks in event handlers or unsubscribing the event handlers.

DivingDuck, 2023-02-12:
@Dardwizzle,

it took some time to sort out all problems around this and some other issues but finally they should be solved now. 

You can test and give us a feedback if you like so. Here is a link to the latest build.
https://github.com/kliment/Printrun/actions/runs/4152400998

I will close the issue and mark as completed

closed 2023-02-12 by DivingDuck