kliment/Printrun #1281: Error When launching

Xeidit, 2022-08-04:
`WARNING:root:RPC server bound on non-default port 7994
Traceback (most recent call last):
  File "/usr/bin/pronterface", line 62, in <module>
    app = PronterApp(False)
  File "/usr/lib/python3/dist-packages/printrun/pronterface.py", line 2439, in __init__
    self.mainwindow = PronterWindow(self)
  File "/usr/lib/python3/dist-packages/printrun/pronterface.py", line 207, in __init__
    self.reload_ui()
  File "/usr/lib/python3/dist-packages/printrun/pronterface.py", line 281, in reload_ui
    self.createGui(self.settings.uimode == _("Compact"),
  File "/usr/lib/python3/dist-packages/printrun/gui/__init__.py", line 259, in createGui
    viz_pane = VizPane(self, vizpanel)
  File "/usr/lib/python3/dist-packages/printrun/gui/viz.py", line 115, in __init__
    root.gwindow = gviz.GvizWindow(build_dimensions = root.build_dimensions_list,
  File "/usr/lib/python3/dist-packages/printrun/gviz.py", line 77, in __init__
    self.p = Gviz(panel, size = size, build_dimensions = build_dimensions, grid = grid, extrusion_width = extrusion_width, bgcolor = bgcolor, realparent = self)
  File "/usr/lib/python3/dist-packages/printrun/gviz.py", line 197, in __init__
    self.mainpen = wx.Pen(wx.Colour(0, 0, 0), penwidth)
TypeError: Pen(): arguments did not match any overloaded call:
  overload 1: too many arguments
  overload 2: argument 2 has unexpected type 'float'
  overload 3: argument 1 has unexpected type 'Colour'

`Version 2.0
Using pop os, 

I'm new with linux so any fixes need to be written for a noob

rockstorm101, 2022-08-06:
Duplicate of #1225.

Hi @Xeidit please see [this comment][1] for the reason behind the error and a temporary workaround. You might need to (re)install Python 3.9. I could provide a .deb file for you if you are into trusting binaries from random strangers on the internet. This is issue seems to be becoming more common and patching through official repositories is not fast enough for it.

[1]: https://github.com/kliment/Printrun/issues/1277#issuecomment-1193904716

rockstorm101, 2023-02-02:
Duplicate of #1225. Which is already fixed. Closing.

closed 2023-02-02 by rockstorm101