kliment/Printrun #1282: Printcore reading temp and storing it in a variable

Bridge313, 2022-08-09:
I need to read the hotend temperature of the printer and store it a variable for later computation. The callback function described in #1023 allows me to send the M105 command but it only prints the temperate to the console.

The way I have figured out how to store the returned output of "M105" is as follows.
I edited printcore.py and added a new attribute to the printcore class called `self.t`. Then I edited the `def _listen():` method and changed line 438 from `try: self.tempcb(line)` to `try: self.t=self.tempcb(line)`.

This way when I use the callback below. The output is stored in `slef.t`
`def temp_callback(a): 
        return a`

Though this code works I am not sure if there is a better way to achieve the result I am after.

volconst, 2022-08-11:
The idea of a callback is printcore to provide temperature to whoever is interested. But it hardly provides any value when it provides a string.
There is no logic printcore to store the result of tempcb it does not use it.
You can store it in your handler.
About the parsing see how pronsole does it
https://github.com/kliment/Printrun/blob/10718fa31c277e35d93b67fc7075aa68cc79da3c/printrun/pronsole.py#L81-L87

tarikcaliskan, 2022-08-11:
Ideally, you can create a temperature queue and consume it one by one.

If you need data instantly you can create and mutate a variable in the callback as @volconst said. That’s how I use printcore.