kliment/Printrun #1283: 3D viewer broken by wxwidgets 3.2/python-wxpython 4.1

ejona86, 2022-08-20:
With wxgtk 3.0/python-wxpython 4.0 the 3D viewer worked fine with printrun 2.0.0rc8. Arch Linux upgraded to 3.2/4.1 and now the GL renderer shows nothing. It appears it is not painting at all. I've attached a screenshot after resizing the window so the lack of drawing is obvious. The 2D viewer works fine, and if I downgrade the wxgtk/python-wxpython version the 3D drawing is restored.

[image]

DivingDuck, 2022-08-22:
Is it possible to update wsPython to the latest version? wxPython 4.2 is official out now and is hopefully able to install on your operating system. Worth a try.

ejona86, 2022-08-22:
wxpython 4.2 hasn't shown up in Arch's testing repo yet, but I figure it will soon. I'll try it out once available.

AlexDaniel, 2022-09-26:
I'm on debian and we have wxpython-tools 4.2.0. It doesn't help:

[image]

Notice that on the screenshot, you see part of the browser window… that's because for me it's not just a black rectangle, but whatever was left on the previous workspace. Still, it's the same bug I think. I'm able to print, it's just that the preview window is not working at all.

j00ch, 2022-10-01:
I tried to debug this using "python3 -m pdb" but i can't find anything wrong. Something must be going awry with stlplater.py

hroncok, 2022-11-18:
See this comment form Fedora: https://bugzilla.redhat.com/show_bug.cgi?id=2140639#c8

closed 2023-06-07 by rockstorm101