kliment/Printrun #1287: M503 crashes Printrun 2.0.0rc8

robertmartin, 2022-09-02:
Running an M503 command crashes pronterface.
printrun.log contents and console contents do not match.
Files attached.
---
Steps to reproduce: 
1. launch pronterface
2. connect to printer
3. issue M503 command
4. press disconnect
---
Result from compiled version, interface crashes. COM port remains open and printer must be manually reset.
Result from source version, errors (see attached) result. Interface is disconnected and will no longer connect. Manual printer reset is required.
M500 and M504 run successfully
Windows 10 Pro (fully updated)
Marlin 1.2.1
Ender 3 Pro
Baud: 114200
[printrun.log](https://github.com/kliment/Printrun/files/9480132/printrun.log)
[pronterface-gui-console.txt](https://github.com/kliment/Printrun/files/9480134/pronterface-gui-console.txt)

volconst, 2022-09-03:
gui log shows macro recursion - a macro you defined which directly or indirectly calls itself.
About the serial timeout - does the printer hang? Can you operate the printer with the lcd after the error?
If the printer hangs on legitimate commands - the problem is in the printer.

FormerCounselor, 2022-09-10:
Hey all. Newbie tech writer here. Just checking my comprehension so I can try to start contributing.  I'm also very new to git. 3D printer knowledge is fair to advanced on a good day. 

The log file is...what the printer view is in pronterface? And the text file seems to be the computer's side of things in pronterface. Looks like the problem is on the Windows side. You say it's the same if done manually...I'd like to see that txt file.

DivingDuck, 2022-10-09:
@robertmartin
did you solve your problem? 
It seems, like @volconst already point to, you have a problem with a macro as shown in your console text file. Maybe it is worth to attach your macro.

Regarding M503 it looks like that your printer loose connection to Pronterface or vise versa. This usually isn't a problem with Pronterface so fare I know from own experiences or had seen in reported issues here. A known cause is using programs in parallel that blocks the com port while running Pronterface with an established connection. Well known from my side is Cura as this program scans all available COM ports with printer commands and lock the port connected of a printer. I disabled the communication plugin in Cura to be save in this regards. Running Putty and Pronterface in parallel on the same port will ask as well for problems if you don't know which program is using the port at a given time. An other cause can be to high communication rates combined with longer USB cables. 
In addition I just recognize in  the first post your baud rate setting: "Baud: 114200". This isn't the usual baud rate setting. 115200 should be the correct standard value. Maybe worth to check too.

robertmartin, 2022-11-02:
I tested again today after upgrading the mainboard on an Ender 3 (to a 4.2.7 version).
The problem remains.

Thank you for your suggestions and insight. Here is some additional information regarding you post.
Cura, PrusaSlicer, Matter Controll, and others never loose connection.
Pronterface is the only application running.
I have tried several different USB cables.
The baud rate was a type-o. I use 115200.
M503 crashes Pronterface both from a macro and manually entering.
M503 from a Putty console works with out issue.

rockstorm101, 2023-07-01:
Hi, I know this is an old issue. @the-robo, did you manage to solve this? I am unable to reproduce this with either Printrun 2.0.0rc8 or the latest 2.0.1 on Debian. Could you try the latest Windows build and see if the problem persists? As was pointed out above the issue might be related to a faulty macro. Maybe try running with a clean configuration file too.

DivingDuck, 2024-06-18:
@rockstorm101 and @volconst,
should we close this issue? I don't see any way to help further.

rockstorm101, 2024-06-19:
Agreed. Last request for information was almost a year ago. Please feel free to re-open if the issue is still there on the latest version.

closed 2024-06-19 by rockstorm101