kliment/Printrun #1288: Nozzle number two

aish7718, 2022-10-08:
Hi, I hope you are well. I can't adjust nozzle number two with pronterface software. Is this software able to turn on the second nozzle?
Thank you for helping me

DivingDuck, 2022-10-08:
Check first via menue:
  -->Settings --> Option --> Printer Settings -->Extruders count matches with the cont of your extruders.

Given that your second extruder is enabled in the firmware you can select the extruder by clicking the Tool button (see 1). Doing so you will recognize the number of the button is changing from 0 to 1. Next time you click on it it will go the the next extruder what is 0 if there are two. Should be two if you have two extruders.
After you select the tool you can use either the heater button (with choosing or input the temperature and hit set, see 2) or hovering with the mouse over the heater gauge and turning the mouse wheel for controlling the temperature. 

[image]

DivingDuck, 2022-10-22:
@aish7718,
had the explanation solve your problem?

DivingDuck, 2023-01-10:
Closed due to missing feedback

closed 2023-01-10 by DivingDuck