kliment/Printrun #1291: 3D view broken in own-compiled package, but the one from Debian repo is good

VanessaE, 2022-11-11:
I gave my PC a much-needed clean-out and reinstall, just to wipe away years' worth of cruft and my own blunders. :smile:  I went from a mixed Debian stable+testing setup to a purely stable install of Debian Bullseye 11.5.0.  The whole reinstall, while time-consuming, went just fine.  By the book.

Of course reinstalling the OS also meant rebuilding and reinstalling my 3d printing toolchain, since underlying libraries will no doubt have changed, but that's where the problem starts.

I can build Pronterface just fine, no obvious errors... compilation Just Works™, though there \*still* isn't a suitable Wx wheel that'll work for Bullseye (need v4.1.0 plus cp39), so the system had to compile it.

The whole build seemed to go without incident:

<details><summary>Show build log...</summary>

```
apt -y install build-essential cmake python3-venv python3-pip \
  python-dbus-dev python3-serial python3-cairo-dev python3-cairosvg \
  python3-cairocffi python-dbus-dev libglib2.0-dev libgtk-3-dev gettext \
  libglu1-mesa-dev python3-opengl python3-pyglet python3-numpy

[apt spew omitted]

vanessa@rainbird:~$ cd /home/vanessa/RepRap/
vanessa@rainbird:~/RepRap$ rm -rf Printrun /home/vanessa/.cache/Printrun/
vanessa@rainbird:~/RepRap$ rm -rf /home/vanessa/.config/Printrun /home/vanessa/.cache/pip

vanessa@rainbird:~/RepRap$ git clone https://github.com/kliment/Printrun.git
Cloning into 'Printrun'...
remote: Enumerating objects: 12625, done.
remote: Counting objects: 100% (140/140), done.
remote: Compressing objects: 100% (88/88), done.
remote: Total 12625 (delta 61), reused 100 (delta 42), pack-reused 12485
Receiving objects: 100% (12625/12625), 80.80 MiB | 23.26 MiB/s, done.
Resolving deltas: 100% (7931/7931), done.
vanessa@rainbird:~/RepRap$ cd Printrun
vanessa@rainbird:~/RepRap/Printrun$ python3 -m venv venv
vanessa@rainbird:~/RepRap/Printrun$ . venv/bin/activate
(venv) vanessa@rainbird:~/RepRap/Printrun$ python -m pip install wheel Cython
Collecting wheel
  Downloading wheel-0.38.4-py3-none-any.whl (36 kB)
Collecting Cython
  Downloading Cython-0.29.32-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.manylinux_2_24_x86_64.whl (2.0 MB)
     |████████████████████████████████| 2.0 MB 4.2 MB/s 
Installing collected packages: wheel, Cython
Successfully installed Cython-0.29.32 wheel-0.38.4
(venv) vanessa@rainbird:~/RepRap/Printrun$ python -m pip install -r requirements.txt
Ignoring pyobjc-framework-Cocoa: markers 'sys_platform == "darwin"' don't match your environment
Ignoring pyreadline: markers 'sys_platform == "win32"' don't match your environment
Collecting pyserial>=3.0
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
     |████████████████████████████████| 90 kB 1.8 MB/s 
Collecting wxPython==4.1.0
  Downloading wxPython-4.1.0.tar.gz (65.8 MB)
     |████████████████████████████████| 65.8 MB 27.1 MB/s 
Collecting numpy>=1.8.2
  Downloading numpy-1.23.4-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (17.1 MB)
     |████████████████████████████████| 17.1 MB 39.8 MB/s 
Collecting pyglet>=1.1
  Downloading pyglet-2.0.0-py3-none-any.whl (966 kB)
     |████████████████████████████████| 966 kB 25.1 MB/s 
Collecting cffi
  Downloading cffi-1.15.1-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (441 kB)
     |████████████████████████████████| 441 kB 36.1 MB/s 
Collecting cairocffi
  Downloading cairocffi-1.4.0.tar.gz (69 kB)
     |████████████████████████████████| 69 kB 7.7 MB/s 
Collecting cairosvg>=1.0.9
  Downloading CairoSVG-2.5.2-py3-none-any.whl (45 kB)
     |████████████████████████████████| 45 kB 3.8 MB/s 
Collecting psutil>=2.1
  Downloading psutil-5.9.4-cp36-abi3-manylinux_2_12_x86_64.manylinux2010_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (280 kB)
     |████████████████████████████████| 280 kB 32.0 MB/s 
Collecting lxml>=2.9.1
  Downloading lxml-4.9.1-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.manylinux_2_24_x86_64.whl (7.0 MB)
[... 66 lines cut ...]
  File "/home/vanessa/RepRap/Printrun/printrun/gui/viz.py", line 69, in __init__
    import printrun.gcview
  File "/home/vanessa/RepRap/Printrun/printrun/gcview.py", line 22, in <module>
    from .gl.panel import wxGLPanel
  File "/home/vanessa/RepRap/Printrun/printrun/gl/panel.py", line 28, in <module>
    from pyglet.gl import glEnable, glDisable, GL_LIGHTING, glLightfv, \
ImportError: cannot import name 'GL_LIGHTING' from 'pyglet.gl' (/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/pyglet/gl/__init__.py)

(pronterface.py:300375): Gtk-CRITICAL **: 23:40:09.089: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:300375): Gtk-CRITICAL **: 23:40:09.089: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:300375): Gtk-CRITICAL **: 23:40:09.090: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
```

(that last message is probably unrelated but I include it here because it says "critical" and repeats about a zillion times)
</details>

Googling around has provided no answers, except for some handwaving blaming `numpy`.

HOWEVER, I also discovered that Pronterface is in Debian's standard repository (didn't notice if it was from `main` or `contrib` though), currently at version 2.0.0-rc8.

I blew away every my Pronterface sources and `apt install`-ed the Debian package.  It works 100% normally.

@rockstorm101 I was advised to ping you about this.

DivingDuck, 2022-11-11:
Hi @VanessaE,
I ran in similar problems last week with my windows version of Pronterface. I thought it was time to change to VS 2022 and all went good up to the moment I build my virtual environment for python 3.10 and ended as well with 2d instead of 3d.

The problem is introduce from pyglet as they have released version 2.0. Unfortunately this version isn't compatible with version 1.xx. I hadn't the time to investigate the source code of panel.py. 

I have a workaround for this that works for windows and I hope this will work for you as well. Change in requirements.txt the line
pyglet (>= 1.1)
to
pyglet >= 1.1, <2.0

This take care that you stick to the latest version 1.xx

Let me know if this work for you.

rockstorm101, 2022-11-12:
> HOWEVER, I also discovered that Pronterface is in Debian's standard repository (didn't notice if it was from `main` or `contrib` though), currently at version 2.0.0-rc8.

Yep, Printrun has been part of the official Debian repositories since early 2016 (it is in 'main', as it is fully free software :) ). However the package is now almost two years old (early 2021) so it probably lacks a few new additions here and there. (nudge intended to push for a 2.0.0 release, got some features in the oven ready for a 2.1.0 release even :P )

> The problem is introduce from pyglet as they have released version 2.0 a couple of days before. Unfortunately this version isn't compatible with version 1.xx.

Quick look at your build and runtime logs, there's clearly something going on with pyglet so @DivingDuck is probably on the right track there. The package currently in stable (a.k.a Bullseye or 11) is using Python 3.9, wxWidgets 4.0 and pyglet 1.5. These versions differ from those on your build log. Let me know if I can help you any further.

VanessaE, 2022-11-12:
@rockstorm101 I would suggest @DivingDuck's fix to `requirements.txt` be updated on the repo rather than in my environment, since we're talking about building under the latest stable release of a major distro. :smile: 

Besides, I'm content to use the Debian package, so I won't need to test building that way myself.

rockstorm101, 2023-02-02:
Note to future self: 8fd9b8b introduced a workaround, issue left open until Printrun supports Pyglet >= 2.0.

real-artswan, 2023-03-22:
Would be good to fix README where it talks about running from sources on Ubuntu/Debian
replace
`pip3 install --user pyglet`
with
`pip3 install --user --force-reinstall -v "pyglet >= 1.1, < 2.0"`

rockstorm101, 2023-03-27:
> Would be good to fix README where it talks about running from sources on Ubuntu/Debian replace `pip3 install --user pyglet` with `pip3 install --user --force-reinstall -v "pyglet >= 1.1, < 2.0"`

Thanks @real-artswan for bringing our attention to this. I'm tempted to even delete the whole [Ubuntu/Debian section][1] there. Users of these distributions probably either use the official package or run from source using a virtual environment. I would say this method using a mix of system and PyPI packages is obsolete and probably should not be encouraged. Did you run Printrun using such method?

[1]: https://github.com/kliment/Printrun#ubuntudebian-1

DivingDuck, 2023-03-29:
It is maybe worth to add requirements.txt in general to the manual build instruction. We we had limit the version beside of pyglet in addition cairosvg and wxpython there as well.