kliment/Printrun #1292: Pronterface no longer connect to Anycubic Kobra Max

Tgwinn, 2022-12-03:
Out of the blue when I try to connect pronterface to my printer it just hangs on "connecting".
-I have no com port errors.
-I do not have Cura running in the background
-Cura still works the printer fine
-I have tried 3 different windows machines, same exact issue, windows finds the printer on Com3 at 9600 no issues
-Pronterface shows correct Com3 at 9600, but just hangs after hitting connect

Any ideas?

Thank you.

DivingDuck, 2022-12-04:
Hi,
will the printer recognize from your operating system when you connect it with your computer? Check if the device manager recognize your printer and the used com port is still the expected port.

volconst, 2022-12-04:
"out of the blue" ... Have you upgraded pronterface or printer firmware in the mean time?
Did the same combination or windows, pronterface, printer versions worked before?
Enable Debug communication from the menu, connect and paste the log here.
Could be that pronterface does not recognize the printer greeting, but either pronterface or printer firmware should have changed.
If you can get similar log (sent, received lines) from cura, include them too.
Does your printer and cura transmit on 9600 baud. Seems too slow.
What is your pronterface version?

Tgwinn, 2022-12-04:
Thanks,
 Yes the printer connects and works just fine. Every computer I've tried sees it on com 3 at 9600, I can run prints from the computer as well.

Tgwinn, 2022-12-04:
-I was using pronterface 1.6.0, so I updated to 2.0.0rc8 (no change, still hangs on connecting)
-The computer is now connecting the printer on Com3 at 57,600
-Cura 5.2.1 connects and operates the printer with no issues
-Selecting debug communications in pronterface keeps just sending M105 over and over with no return result
-No changes I'm aware of

*I don't see a way to debug or get a coms report in Cura?

Thank you for your responses.

DivingDuck, 2022-12-04:
Try with baud rate 115300.
In addition, as you maybe jump between 1.6 and 2.x versions, rename the file printrunconf.ini to printrunconf.ini.old
The configuration file between both version isn't compatible and will damage the setup. Pronterface will create a new configuration once it do not find a file.
Attention, there are two locations for Windows, one is in your user directory at 
`c:\user\YourUsername\printrunconf.ini`
and the other location is at
`c:\Users\YourUsername\AppData\Local\Printrun\Printrun\printrunconf.ini`
You need to rename/delete both. After starting Pronterface go to Settings and adjust your setup.

Activate logging in Pronterface. This may help to find out what is going on. Here is an example. Set your com port and baud rate as well as your bed size (Printer settings) and in the next tab (User interface) the location and name for your logging file, like `c:\user\YourUsername\printrun.log`.

[image]
[image]

Tgwinn, 2022-12-04:
@DivingDuck heck yeah!!!! 115200 baud rate connected!!!!
Rock solid support sir, thank you!!

Tgwinn, 2022-12-04:
Many tanks to the community, great support and I learned a good bit.

closed 2022-12-04 by Tgwinn

DivingDuck, 2022-12-04:
You are welcome. Happy printing.