kliment/Printrun #1293: connection issues

Bapp1, 2022-12-18:
When i open pronterface it says i have com port 1 and it is set to 15200 but click connect and it just sits there saying connecting.

DivingDuck, 2022-12-18:
Sure that you use COM1 for connecting to your printer? This port is usually used by a mouse etc.
A adequate baud for a connected printer is usually higher, try 115200.
[image]

Bapp1, 2022-12-18:
thats the only com port i see and i dont know why it wont connect up both mouse and keyboard work from bluetooth.

[image]
[image]

DivingDuck, 2022-12-19:
Sorry, I missed your answer. I'm quite sure that this port isn't the port your printer is using. At least this is not very common. It is maybe worth to check your device manager what happen when you connect your printer:

Close Pronterface and any slicing software. Disconnect your printer from the computer. Open your device manager (Keyboard  Windows+X and select device manager) and look to your COM port entries. There should be nothing listed if there is nothing connected, but for most users there is COM1 listed as communication port - so COM1 is usually the only port you see in the listing. 
Now connect your 3d printer. You should see a new entry for your printer in the device manager (and you should see and hear in addition from Windows a notification that a new device was connected). 
If you do not see a new entry for your printer this means that Windows do not recognize the printer. This happen if Windows did not find a driver for your printer motherboard. You then need to find a adequate driver from your printer manufacturer and install it. Usually Windows should find the driver itself, but sometimes it can't  and you need to help manually.

And keep in mind that some slicers will lock COM ports if they find a 3d-printer. No other program then is able to connect to this port in parallel. For example Cura has such a behavior, so don't use them at the same time

Bapp1, 2022-12-19:
OK i removed the com port did a scan for changes it reloaded com 1 updated drivers and plugged printer in nothing showed up via usb or com ports i downloaded CH341SER.zip which is supposed to be the drivers i need and not CDM v2.12.28 WHQL Certified. I had previously installed them. No 3d slicer programs or otherwise were loaded or running. Opened pronter and again it shows com 1 but doesnt connect, not sure how else to sort this, Any ideas ??

DivingDuck, 2022-12-20:
What happen in the device manager when you disconnect your printer?

I should have ask this question before: What version of Pronterface are you running and what kind of printer hardware / controller board?
You mentioned you use a CH341SER driver. You should see at least this driver in your COM port listing when you connect your printer and every thing with the driver installation went well. 
In addition you can check this as well when you click on show hidden devices in the Device Manager View menu. This will show you not connected devises.

Bapp1, 2022-12-20:
Sorry i thought id put it in already 
printer is an Ender 3 with a btt skr mini  3e v3.0 main board with a btt tft 35 touch screen and im running the latest version of pronterface-windows-x64_3.8-2.0.0rc8 running on windows 10 pro i also have a pi3b with octopi-0.18.0-1.8.1 installed but i cant get that connected at the mo either seems having to reinstall upset a few things. I have discovered last night however that when plugging in pronter i get Unknown USB Device Descriptor Request Failed now having gone down through a check list of usb fixes

DivingDuck, 2022-12-20:
So at least your OS recognize a new device. Now you need to find the correct device driver for your board. Try to find them with Windows Update. Right click the unknown device and let windows search for the driver.

Bapp1, 2022-12-20:
done that and all updates done also have  drive booster that checks for all
the updated drivers and thats done as well just got to work out the kinks
and hopefully have it sorted.

DivingDuck, 2022-12-20:
Great, so then happy experimenting & printing.
Please close the issue as soon as everything works fine with Pronterface for you.

DivingDuck, 2023-02-12:
I close the issue as solved.

closed 2023-02-12 by DivingDuck

Bapp1, 2023-02-12:
thought id done that already pal  but thanks