kliment/Printrun #1294: Does printrun support SLA printer

arslan437, 2022-12-28:
I looking for a opensource software for SLA printer. Some have mentioned Printrun https://3dprinting.stackexchange.com/questions/1217/how-to-display-images-on-dlp-using-hdmi-for-3d-printing

My plan is to make a windows based software with the following features

- It talks to Marlin based controller to move the z axis. 
- It controls to displays. One is the user screen with the gui application and the other one will be the UV project that will project the sliced pattern on platform. 

Does printrun can provide any help with this kind of software ?

rockstorm101, 2023-01-23:
Hi @arslan437, sorry for such a late response. I don't own an SLA (or DLP) printer so I can't really tell whether they are really usable with Printrun. Printrun was developed more towards extrusion printers and therefore expects hot ends and the like. However, Pronterface/Pronsole are both able to talk Marlin firmware and send it any G-code commands you wish. Displaying exactly what you are after, I highly doubt it.

I would suggest trying printing a G-code file with Pronsole. It might do the trick just fine if Marlin is what's listening on the other end. If not, I'd be happy to accept changes towards making Pronsole able to handle SLA printers as well. For Pronterface it sounds like it will require quite some rework to be honest.

volconst, 2023-01-23:
@arslan437 seems to work https://github.com/kliment/Printrun/issues/1140#issuecomment-756978906

This is the SLA GUI https://github.com/kliment/Printrun/blob/8fd9b8b6dab86ee0c238016f9a0ed7b0c5abea73/printrun/projectlayer.py#L819
It should be able to start from python command line:
`python3 -m printrun.projectlayer` or sth
I think it can also be started from the menu.
I do not think it is usable.

DivingDuck, 2023-01-23:
For Windows it will not run out of the box with the actual version. The projector feature is related on GTK3 and those are not available in general for Windows. I did the integration for the needed libraries here:  #1169 or you can test this here: https://github.com/DivingDuck/Printrun/actions/runs/3748727767. 

I agree to @volconst and @rockstorm101

arslan437, 2023-01-24:
So I have an idea to make an DLP/SLA printer without any raspberry pi or SBC. 

### With the following features

- Easy to use 
- Load the stl and just press the print(All the slicing and other stuff will be done by the software it self)
- Initially hardware is to controlled by the marlin. Later to develop a new or strip the marlin to custom firmware. 
- Software to run on windows (because mostly people use that)
- it will detect the connected display with the computer 
- When user press the start button it will slice the file and home the machine than display the image/pattent on the LCD screen/project to cure the resin and move z axis and so on.  

### Currently struggling at 

- At this time my plan is to use the python to build the gui and other stuff. Because it will be much faster to build. 
- I don't want to build the slicer. looking for a Command line slicer. Maybe pursa will work. But don't know if we can change the build plate size for sla printers or not. 
- I have figured out that .sl1 file is basically a zip file.  

I am looking for your idea and if any one have any suggestion on it. Do you guys think is it going to be a good project or not. If I can do this then the SLA/DLP printers will much easier to build as compared to FDM printers.

arslan437, 2023-01-24:
I didn't got any response from here so did started to work on my own software.  
I am able to display the images on the second display on the windows using pyglet.
don't know how to disable the second display when computer is powered on. How to stop windows to not use that display.

arslan437, 2023-01-24:
I have tested your compiled version it does open a new. I am able to open the sliced file on the website that you linked. but getting this error 
I am using the exe_windows-latest_x64_3.10
```
Layer thickness detected:Traceback (most recent call last):
```

DivingDuck, 2023-01-24:
Is there more information? I did not get this error.
Here is how it looks on my PC with an example file (see attached svg demo file):

1. Open Projector in Pronterface

[image]

2. Load SVG demo file (I sliced it with Slic3r)

[image]

3. Run the presentation 

[image]

Sample file:
[Drehmeiselbrett v3-exp.zip](https://github.com/kliment/Printrun/files/10487830/Drehmeiselbrett.v3-exp.zip)