kliment/Printrun #1295: WIn-build Actions are expired - no downloads of any working builds that newer than 2.0.0rc8

DaLiV, 2023-01-08:
wanted to try latest versions for windows
- latest throw exceptions
- but all other previous builds are downloads "expired"

DivingDuck, 2023-01-08:
Had you try this run?
https://github.com/kliment/Printrun/actions/runs/3682013645

Development versions from actions are only a limited time available and do expire.

DivingDuck, 2023-01-10:
Thanks for this info. This this seems to be the wxPython issue and happen to Windows 10 and 11 with some language setting combinations. I have a version where this is solved. You can try this interim version if you want:

https://github.com/DivingDuck/Printrun/actions/runs/3748727767

Scroll to the end of the page and download one of the artifacts. The number at the end of the file identify the python version used for compiling. This is the only difference.

This Wiki page list all changes compared to  the latest release candidate:
https://github.com/DivingDuck/Printrun/wiki/My-latest-development-version-build-with-github-workflow
Feedback is welcome.

DaLiV, 2023-01-10:
gui has started but on [connect] - not connected ... and no problem descriptions - empty traceback.
```
Connecting...
--- Logging error ---
Traceback (most recent call last):

During handling of the above exception, another exception occurred:
```

+ On Close (must terminate in task manager) :

```
Not connected to printer.
--- Logging error ---
Traceback (most recent call last):
  File "logging\__init__.py", line 1088, in emit
  File "printrun\pronterface.py", line 108, in write
AttributeError: 'NoneType' object has no attribute 'write'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "printrun\printcore.py", line 150, in logError
  File "printrun\pronsole.py", line 297, in logError
  File "logging\__init__.py", line 2049, in error
  File "logging\__init__.py", line 1475, in error
  File "logging\__init__.py", line 1589, in _log
  File "logging\__init__.py", line 1599, in handle
  File "logging\__init__.py", line 1661, in callHandlers
  File "logging\__init__.py", line 954, in handle
  File "logging\__init__.py", line 1093, in emit
  File "logging\__init__.py", line 1006, in handleError
  File "printrun\pronterface.py", line 108, in write
AttributeError: 'NoneType' object has no attribute 'write'

--- Logging error ---
Traceback (most recent call last):
During handling of the above exception, another exception occurred:
During handling of the above exception, another exception occurred:
During handling of the above exception, another exception occurred:
```

DivingDuck, 2023-01-10:
Interesting, you are the the second person reporting this. Can you check the issue below?
https://github.com/DivingDuck/Printrun/issues/22
It seems like you have a defect printrunconf.ini. This happen usually if you switch between 1x and 2.x versions. 

Please delete this file in: 
`c:\Users\Your_User_Name\`
and delete the folder:
`c:\Users\Your_User_Name\AppData\Local\Printrun\`
as described. 
Then open Pronterface and set the log file to an appropriate file name including the path information as I had shown. You need to have full read/write access to this location. Like:
`c:\Users\Your_User_Name\Printrun.log`  

Your_User_Name = your windows user name

DaLiV, 2023-01-10:
so ... that was "log file" issue
tried - ok,
but after
- restored previous config from 2.0.0rc8
- changed log location
- killed (as tracebacks and not closing)
= working

to solve / improve : 
a) default = not to log, especially when no log file set (that must be advisable when software try to do that)
b) set default log location in %TEMP%, %LOCALAPPDATA%, %APPDATA% ... not in %USERPROFILE%
c) on unwriteable log - alert and open "create LOG file" dialog

DivingDuck, 2023-01-10:
Thanks for your feedback, I appreciate that. 
The original behavior was not to log if there is no logging file specified. 
I'll take a look on that but I can't promise to change this as we do not have only windows users to consider here. In addition %LOCALAPPDATA%, %APPDATA% are hidden folders and usual users do not to see this locations in Explorer what cause additional problems / issue reports.

Anyway, I see the behavior of Pronterface had change and I can reproduce the issue with your information. There should be at least a check to prevent logging if there is no entry made.

Again, thanks for reporting.

DivingDuck, 2023-01-11:
I had the chance to take a quick look on the issue. Pronterface should redirect all logging activities to console output if there is no log path specified.
Exact this happen if I run Pronterface directly from source code. Running Ponterface from the executable file will show an error if log path is not set.
This happen in the latest artifact builds here as well as in my latest builds local and via my repository. 
I have no solution for this right now. 
The work around for now is to set path and filename in log path to an appropriate location where the user have read/write access. Like: 
`c:\Users\Your_User_Name\Printrun.log` 
(Your_User_Name = your windows user name)

DivingDuck, 2023-02-12:
@DaLiV ,
it took a while to fix that problem as well, but now this issue is solved too with the latest updates we did for the repository here. You don't need to download the files from my repository as it is in sync with my updates. You can test and give us a feedback  if you like so. Here is a link to the latest build. 
https://github.com/kliment/Printrun/actions/runs/4152400998

I will close the issue and mark as completed (https://github.com/kliment/Printrun/commit/e66bbbdeab17dfd6951299d92c4d8dc0cd32bb05)

PR #1308 (2023-03-15): Release Printrun 2.0.0
Now that #928, #1171 have been dealt with and Windows side has been revamped (many thanks @DivingDuck) I believe we are in shape for finally releasing 2.0.0. Up for debate of course, please speak up if you know any show-stoppers.

Two things I can think of that'd be nice to do before releasing:
 - [x] Test and confirm macOS binaries work as expected
 - [x] Revisit Windows installation instructions and update (if required)

--
Fixes #1305
Fixes #1296
Fixes #1295
Fixes #1252
closed 2023-02-12 by DivingDuck