kliment/Printrun #1297: It should be possible to switch autocapitalization on and off. I think this is a due enhancement.

FrHePo, 2023-01-15:


rockstorm101, 2023-01-16:
Hi, would you please be able to provide an example of when this causes an issue?

FrHePo, 2023-01-16:
https://github.com/prusa3d/Prusa-Firmware/issues/3891

wavexx, 2023-01-23:
@rockstorm101 some commands expect data to pass though as-is. The dumbest command I can think of that expects data to pass though is setting an LCD message. In the prusa FW the memory read/write commands expect to contain lower-case hex characters. We can adjust the parser, but that will break all old firmwares.

See https://github.com/kliment/Printrun/pull/1298 for a fix proposal which implements what is written here https://github.com/kliment/Printrun/issues/216#issuecomment-18603449