kliment/Printrun #1300: release_windows.bat produces broken pronterface.exe

vnd, 2023-01-23:
I'm building vanilla* Printrun (latest master), however when I open the resulting pronterface.exe and immediately try to close it (no 3D printing, no other actions at all), there's the following error printed and closing window doesn't work:
```
Not connected to printer.
--- Logging error ---
Traceback (most recent call last):
  File "logging\__init__.py", line 1028, in emit
  File "printrun\pronterface.py", line 108, in write
AttributeError: 'NoneType' object has no attribute 'write'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "printrun\printcore.py", line 150, in logError
  File "printrun\pronsole.py", line 297, in logError
  File "logging\__init__.py", line 1959, in error
  File "logging\__init__.py", line 1407, in error
  File "logging\__init__.py", line 1514, in _log
  File "logging\__init__.py", line 1524, in handle
  File "logging\__init__.py", line 1586, in callHandlers
  File "logging\__init__.py", line 894, in handle
  File "logging\__init__.py", line 1033, in emit
  File "logging\__init__.py", line 946, in handleError
  File "printrun\pronterface.py", line 108, in write
AttributeError: 'NoneType' object has no attribute 'write'

--- Logging error ---
Traceback (most recent call last):

During handling of the above exception, another exception occurred:
During handling of the above exception, another exception occurred:
During handling of the above exception, another exception occurred:
```

* the only difference is commented out line "pip install --upgrade pip", b/c if not commented out it uninstalls pip, then installing it again fails with "Access is denied" error. However the same error is also visible later in build logs when script tries to update all packages.

release_windows.bat log:
https://pastebin.com/ZYd8YeU6

Windows 11, python 3.7.8

DivingDuck, 2023-01-23:
This version is outdated and have several little problems with Windows 10/11. In addition it is maybe a good idea to switch to Python .310

For the meantime you may try to use the version of my repository. I did some fixes regarding known problems, add some pull requests and adjustments related to Python 3.10 and wxPython 4.2. All differences and the link to the artifact files (on the bottom of the wiki page) are listed here:

https://github.com/DivingDuck/Printrun/wiki/My-latest-development-version-build-with-github-workflow

For cloning:
Switch to branch [testPython310], it holds all changes listed in the wiki mentioned above. 
In case you want to run the projector feature you need to add the submodul [PrintrunGTK3](https://github.com/DivingDuck/PrintrunGTK3.git) as well to the project (mentioned in the wiki as well).

vnd, 2023-01-24:
@DivingDuck Thanks for your suggestion, but unfortunately it didn't help -- I've installed python 3.10, built testPython3.10 branch but still getting the same error. I was thinking that may be it's my environment problem, but reinstalling C compiler environment didn't help either, so not sure what could be problem. Overall it's pretty much stock Windows 11 virtual machine, without much customization.

Build log: pastebin.com/eBGbMB91

DivingDuck, 2023-01-24:
I see two things:

1. --- Logging error ---
For this exist a workaround that I add yesterday in the wiki: https://github.com/DivingDuck/Printrun/wiki/My-latest-development-version-build-with-github-workflow (see last entry at the bottom). Set the logging path manually to an exiting location

2. Cairo library not found - cairosvg will likely fail to work
Yea, there is something wrong in your v3 environment. See line 228 to 230 of your build log: The Cairo library is missing. This should be installed from the requirements.txt (see batch file release_windows.bat line 107. Delete the folder v3 and run release_windows.bat once more. You need to have an active internet connection so that pip is able to download / update the needed libraries

In addition I saw:

3. Pyglet 2.0 error:  Your graphics drivers do not support OpenGL 2.0.
Only a remark, your computer graphic driver or card seems to be outdated. Should not be a problem as I downgrade this version to 1.5.x

4. It looks like the submodule PrintrunGTK3 is missing too
This can be added in two ways. Clone the repository https://github.com/DivingDuck/PrintrunGTK3 and copy it to directory C:\DivingDuck\Printrun or use git command line  git submodule within C:\DivingDuck\Printrun:

    `git submodule add https://github.com/DivingDuck/PrintrunGTK3`

    `git submodule update --init --recursive`

This will clone the repository into the project folder. The second command is used for updating the submodule in case I release a new version. The repository provides some needed GTK3 library files used for the projector feature used in projectlayer.py

vnd, 2023-01-24:
Thank you, somehow I've overlooked logging error info in the notes, however I caught it also with kliment/Printrun.

Regarding Cairo -- did a clean rebuild (deleted v3), and still got the same logging regarding it. But not sure it actually has any user-visible side-effects so all good.

vnd, 2023-01-26:
Overall testPython3.10 branch actually works when it comes to pronterface.exe generation. With master from this repo I was constantly experiencing bizarre bugs, not seen when running pronterface.py (it's a printrun fork with some more dependent libraries).

DivingDuck, 2023-01-26:
That's correct and intended as I was waiting for a pull request. The branch testPython310 is my actual branch. It includes corrections regarding wxPython, pyglet and things I forgot already (but listed in the wiki page ). In addition it includes a separate windows action chain for both test- and testPython310 branch. 
I will update the wiki page when I make updated versions. This happen mainly if there is a problem with new python or library versions that need some minor adjustments.

I don't want to establish a second alternative version of this repository here. My repository is only a snapshot of what I am using and is tested for Windows only. Usable as a temporary solution.

The original version here is the one and only main repository. (This is why I do not want to make own releases)

rockstorm101, 2023-02-02:
Fix was merged in 0c296ba so I'm closing this issue. Please do re-open if I got that wrong.

vnd, 2023-02-10:
The issue is still there, exactly as described in the first comment. Unless I missed something in @DivingDuck commit log, there's no fix for it, only a workaround to set 'Log path' to a writable location.

DivingDuck, 2023-02-11:
Yes, the logging issue is still there, thanks for the reminder, Sorry, I forgot this one. I reopen the issue and close it when I'm done with the correction (hopefully today).

It turns out this wasn't a problem with Pronterface or with release_windows.bat but with the actual version of pyInstaller 5.7.0. They change it's behavior for at least windows operating systems in a way that it breaks the code as you and two other user had recognize in their installation. I wonder if this happen for OSX as well. 
[Reference to pyInstaller 5.7.0changes](https://pyinstaller.org/en/stable/CHANGES.html#incompatible-changes)

The generation of windows binaries is working now for the branch here since we merged in all my changes here. With the workaround it should work for now.

Anyway I nailed down the problem yesterday evening and there will be an update from me soon.

@rockstorm101, I will refer to the issue here for the update.

DivingDuck, 2023-02-11:
This issue should be solved as soon as pull request #1311 is approved. In the meantime you can try the very latest binary and let us know how it works for you.
[Windows artifact:](https://github.com/kliment/Printrun/suites/10921684808/artifacts/552118766)

commit e66bbbde (2023-02-11): Update translation templates, German translation, fix logging path on Windows

* Update translation master templates for 2.0.0
* Updated German translation files for 2.0.0
* Remove binary .mo files
add *.mo and plater.spec to .gitignore
* Restrict windows build done to Python 3.10 only
* Implement fallback for invalid logging path that crashes the windows application due do a change in pyInstaller v5.7.0
fixes #1300
closed 2023-02-11 by kliment