kliment/Printrun #1302: Pronterface GUI becomes unresponsive while printing, need to kill the app

hroncok, 2023-01-26:
Hey folks. I've finally dusted off my Lulzbot Mini and decided to use the Fedora-packaged Pronterface on Fedora 37 (2.0.0rc8 with #1224 and #1262, Python 3.11.1, wxpython 4.2.0, pyserial 3.5, Xfce and hence X11 and not Wayland).

Pronterface allows me to connect to the printer, manually control it and start the print.
After several minutes of printing, while working on other stuff when the Pronterface window is not visible on my desktop when I switch back, the GUI is unresponsive -- in fact, the window is "empty" (it keeps the content of whatever was on the screen before I switched to it). The print however goes on. When the print is finished, the app remains unresponsive.

The terminal I used to start the Pronterface is riddled with *Attempted to write invalid text to console, which could be due to an invalid baudrate* (but my baudrate is OK).

Considering this is printed on *any* exception, I suspect something is happening there:

https://github.com/kliment/Printrun/blob/8fd9b8b6dab86ee0c238016f9a0ed7b0c5abea73/printrun/pronterface.py#L778-L779

I will log the actual exception and see if I debug this further. I have a strong suspicion this will turn out to be yet another float/int issue.

<details>
<summary>My cat looking at the ongoing print (not really relevant)</summary>

[image]

</details>

hroncok, 2023-01-26:
As expected:

```
Traceback (most recent call last):
  File "/usr/lib64/python3.11/site-packages/printrun/pronterface.py", line 763, in addtexttolog
    self.logbox.Remove(0, current_length / 10)
TypeError: TextEntry.Remove(): argument 2 has unexpected type 'float'
```

DivingDuck, 2023-01-26:
Interesting, I don't see it for Python 3.10 on windows. I plan to switch to Python 3.11, so I should better look for it too.

hroncok, 2023-01-26:
The console needs to be visible and the number of lines (?) already printed must be > 20000.

My Luzbot Mini spams the console with messages like:

    ok N16224 P0 B3

...a lot.

I don't think this happens often.

closed 2023-01-26 by kliment