kliment/Printrun #1304: RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted

rockstorm101, 2023-01-26:
### Steps to reproduce

 - Checkout latest master https://github.com/kliment/Printrun/commit/8fd9b8b6dab86ee0c238016f9a0ed7b0c5abea73
 - Fresh install (i.e. delete all 'pronsolerc' files lying around)
 - Run `./pronterface.py`
 - Settings -> Options -> Viewer -> Main visualization -> Change from '3D' to '2D'

### Error

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/wx/core.py", line 3427, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rock/Projects/printrun/rockintrun/printrun/gl/panel.py", line 139, in Refresh
    return super().Refresh(eraseback)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
```

### System Info

 - Debian 12
 - Python 3.11.1
 - wxPython 4.2.0
 - Pyglet 1.5.27

rockstorm101, 2023-01-27:
> Funnily enough only without `~/.pronsolerc`. When I try to get a new one, no matter what I do in the app, the file is not created.

Isn't that expected? When there is no '~/.pronsolerc' the new configuration file is created at '~/.config/Printrun/pronsolerc' instead.

hroncok, 2023-01-27:
Aha!