kliment/Printrun #1305: UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 12: ordinal not in range(128)

rociomarielgnas, 2023-01-29:
Hello.
I have been trying to run pronterface for a while now on Windows 10, I have the lastest binaries from the file Printrun-win-18Nov2017.zip but I had no luck. I get this message:

[image]

This is what the log prints each time:

```
Traceback (most recent call last):
  File "pronterface.py", line 66, in <module>
  File "printrun\pronterface.pyc", line 2294, in __init__
  File "wx\_core.pyc", line 7974, in __init__
  File "wx\_misc.pyc", line 6216, in SetInstallPrefix
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 12: ordinal not in range(128)
```

I tried deleting the ini config file from my AppData folder but that didn't change anything.

Thanks in advance!

rockstorm101, 2023-01-29:
> I have the lastest binaries from the file Printrun-win-18Nov2017.zip

That sounds like quite old binaries. Please try and let us know if the problem persists:
 - [Printrun 2.0.0-rc8](https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc8)
 - ot the latest [nightly build](https://github.com/kliment/Printrun/suites/10599271621/artifacts/528359928)

rociomarielgnas, 2023-01-29:
Hi, thank you for replying so quickly!
I extracted and ran the exe inside the pronterface-windows-x64_3.8-2.0.0rc8 file and got this message:
[image]
This time no log file was generated. There was a new ini config file on AppData but it was empty.

Running the nightly build yielded this message:
[image]

```
Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
  File "printrun\pronterface.py", line 2582, in __init__
  File "printrun\pronterface.py", line 186, in __init__
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1699) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!
```
My computer's region is Argentina in case it helps.

rockstorm101, 2023-01-31:
Hi @rociomarielgnas, sorry for the delay in answering, please have a try with the very [latest binary][1] and let us know how that goes.

[1]: https://github.com/kliment/Printrun/actions/runs/4058589985

rociomarielgnas, 2023-02-01:
> Hi @rociomarielgnas, sorry for the delay in answering, please have a try with the very [latest binary](https://github.com/kliment/Printrun/actions/runs/4058589985) and let us know how that goes.

It finally opened!! Thank you so much!

PR #1308 (2023-03-15): Release Printrun 2.0.0
Now that #928, #1171 have been dealt with and Windows side has been revamped (many thanks @DivingDuck) I believe we are in shape for finally releasing 2.0.0. Up for debate of course, please speak up if you know any show-stoppers.

Two things I can think of that'd be nice to do before releasing:
 - [x] Test and confirm macOS binaries work as expected
 - [x] Revisit Windows installation instructions and update (if required)

--
Fixes #1305
Fixes #1296
Fixes #1295
Fixes #1252
closed 2023-03-15 by rockstorm101