kliment/Printrun #1312: trojan?

Jitone1, 2023-02-07:
[pronterface-windows-x64_3.8-2.0.0rc8.zip](https://github.com/kliment/Printrun/releases/download/printrun-2.0.0rc8/pronterface-windows-x64_3.8-2.0.0rc8.zip) 
VirusTotal comes back with:
 Cyren
W64/S-a720cd15!Eldorado
Google
Detected

Paul-PADataAnalytics, 2023-03-15:
Virus reported on the latest build.
https://github.com/kliment/Printrun/actions/runs/4363280108

Google and Windows Defender report this.  Windows defender reports Trojan:Win32/Wacatac.H!ml

rockstorm101, 2023-03-27:
@Jitone1, @Paul-PADataAnalytics, is this still an issue with the [latest release][1]?

[1]: https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0

pansito, 2024-09-15:
[[printrun-2.1.0_windows_x64_py3.10.zip](https://github.com/kliment/Printrun/releases/download/printrun-2.1.0/printrun-2.1.0_windows_x64_py3.10.zip)](https://github.com/kliment/Printrun/releases/download/printrun-2.1.0/printrun-2.1.0_windows_x64_py3.10.zip)
 The download is interrupted by my browser due to a virus detection. 
[image]

DivingDuck, 2024-09-21:
@pansito,
thanks for your information.
Pronterface applications are not signed by Microsoft as this is an community project. Your browser did a false positive detection as already mentioned. You should report this to your browser support team.

Craftingdaemon, 2026-10-15:
Windows defender detected "Trojan:Win32/Vigorf.A"