kliment/Printrun #1315: Gcode view: 2D mode crashing on fast layer change

swiftgeek, 2023-02-10:
```
Traceback (most recent call last):
  File "/home/swiftgeek/tmp/Printrun-upstream/printrun/gviz.py", line 405, in paint
    self.repaint_everything()
  File "/home/swiftgeek/tmp/Printrun-upstream/printrun/gviz.py", line 331, in repaint_everything
    dc.SelectObject(self.blitmap)
wx._core.wxAssertionError: C++ assertion "cr && cairo_status(cr) == 0" failed at /usr/src/debug/wxwidgets/wxWidgets-3.2.1/src/gtk/bitmap.cpp(1399) in CairoCreate(): 
zsh: terminated  python ./pronterface.py
```

First it stops rendering while displaying traceback above after attempt to change layer, and after a bit more layer change attempts crashes entire pronterface

rockstorm101, 2023-02-15:
Hi @swiftgeek I'm unable to replicate this, does it occur with a specific .gcode file? Could you provide it? Or please be more specific on how to reproduce it

swiftgeek, 2023-02-15:
No specific file, simply changing layer fast with mouse wheel. Changing one layer at a time and letting it render before changing again doesn't break anything

rockstorm101, 2023-02-15:
I'm unable to replicate this then. On the "Gcode view" tool, scrolling as fast as my mouse wheel allows and changing zoom level using Ctrl+Mouse Wheel does not produce the above either.

Out of curiosity:
 - this happens even with #1314 applied?
 - what version of python gobject libraries (`python3-gi` on Debian/Ubuntu) are you using?

--
Debian Sid
Python 3.11.1
wxPython 4.2.0

swiftgeek, 2023-02-15:
* #1314 has no effect
* python-gobject [3.42.2](https://gitlab.gnome.org/GNOME/pygobject/-/tags/3.42.2)