kliment/Printrun #1316: Error connecting via usb com port

Blackcatswiskas, 2023-02-15:
I am getting the following message on trying to connect 
_

> Could not connect to COM4 at baudrate 115200:
> Serial error: could not open port 'COM4': PermissionError(13, 'Access is denied.', None, 5)

_
I have tried changing ports, baud rates, clearing usb cached info in device manager and different cables
I have a makerbase Robin nano board installed in an Xvico 3xs 

Can someone please help I need to sort my PID calibration asap

DivingDuck, 2023-02-15:
Permission error usually indicate that you have locked the Port with an other application. Like running Cura in parallel. In addition check if COM4 is the port you use for the connected printer.

Blackcatswiskas, 2023-02-15:
I have checked all the ports are empty except my keyboard and mouse.  I have cleared the hidden usb entries in system on control panel 

>

DivingDuck, 2023-02-17:
Hi,
 
did you check the firmware setup? MKS often use a faster baud rate (250000) in their default setup.

You can try to use a terminal app like Putty in case you don't got it running but usually Pronterface is quiet good in serial connection.

BTW, with what version of Pronterface are you working?

Blackcatswiskas, 2023-05-16:
Sorry All Sorted port was indeed locked

closed 2023-05-16 by Blackcatswiskas