kliment/Printrun #1318: Enable the --use-pep517 flag on build workflows

rockstorm101, 2023-03-06:
Deprecation warning during builds:
```
DEPRECATION: cairocffi is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
```

See [this comment][1] for more context.

[1]: https://github.com/kliment/Printrun/pull/1317#issuecomment-1455042360

rockstorm101, 2024-01-26:
Dependencies `cffi`, `cairocffi`, `cairosvg` were removed as of 556fe07 and these warning are no longer present on build logs. I think we can close this issue?

DivingDuck, 2024-01-26:
Yes, we can close this issue. We removed them all with #1400 and #1387.

closed 2024-01-27 by rockstorm101