kliment/Printrun #1321: Can't set locale Macbook with macOS 12.6.3

DivingDuck, 2023-03-07:
Follow up for issue https://github.com/kliment/Printrun/pull/1308#issuecomment-1455217937,

This need some more investigation as
a) the language combination is wrong,
b) we use from the coding eg. de-AT only the first part for identifying the language, so it should find a fallback,
c) I'm not sure if the warning is from Pronterface, I can't even find any message string for this window. Strange.

Check if the implemented workaround issue #1154 is still needed for macOS builds

@neofelis2X,
can you please add the log file from Pronterface here? Maybe this will show us some more useful information. 
For creating the logging file you need to set up a log path in: -->Settings -->Options -->User interface -->Log path
It must be an existing folder where you have full access to. The log file name will be set from Pronterface: Printrun.log

[image]

neofelis2X, 2023-03-07:
There is nothing in the logfile, unfortunately.
>2023-03-07 21:08:02,888 - Not connected to printer.
2023-03-07 21:08:02,889 - Disconnecting from printer...
2023-03-07 21:08:02,889 - Exiting program. Goodbye!
>

One comment on a): As far as my computer is concerned, the language combination is not wrong or incorrect. In fact, it's just how I set it up. System language in English and region is Austria. Gives 'en_AT'. I know that this is a highly unusual combination, but on the mac it is a perfectly good setting to have. The issue might be an edge-case tho. :)

I wrote a python script to test a few different methods. The result is that most likely `locale.setlocale()` (util.py line 34) cannot digest this unusual locale. When I set en_GB  for example, it runs just fine.
>wx.Locale.GetSystemLanguage:  180
wx.Locale.GetLanguageCanonicalName:  en_AT
locale.getdefaultlocale:  ('en_AT', 'UTF-8')
os.getenv:  en_AT.UTF-8
locale.setlocale...
Traceback (most recent call last):
  File "/.../Printrun/locale_test.py", line 22, in <module>
    locale.setlocale(locale.LC_ALL, '')
  File "/.../python3.10/locale.py", line 620, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
>

And on a side note, `locale.getdefaultlocale()` in utils.py will be deprecated in py3.11. But it's not the problem here. :)

rockstorm101, 2023-03-07:
> One comment on a): As far as my computer is concerned, the language combination is not wrong or incorrect. In fact, it's just how I set it up. System language in English and region is Austria. Gives 'en_AT'. I know that this is a highly unusual combination

I agree, it is unusual but not impossible. As @DivingDuck said on his point b), we should be able to fallback to a sensible default in these cases.