kliment/Printrun #1332: Exchange of module pyreadline with pyreadline3 in requirements.txt

DivingDuck, 2023-03-29:
The module pyreadline seams to have no maintenance. By reviewing our latest latest build log I recognize a note in https://github.com/kliment/Printrun/actions/runs/4542190897/jobs/8005293012, line 107:

`DEPRECATION: pyreadline is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559`

There exists a new fork called pyreadline3 that seems to be maintained and solved some of the problems the original module had and solve as well the deprecation note.

**Reference**

**pyreadline**: https://github.com/pyreadline/pyreadline/issues/76

**pyreadline3**: https://github.com/pyreadline3/pyreadline3

I integrated the new module in my repository for a couple of live prints and there was no problems so fare. So my suggestion is to switch to the new module.

Related to Windows build only

Please let me know what you are think about this @kliment, @rockstorm101

rockstorm101, 2023-03-30:
Hi @DivingDuck, did you run your tests on pronterface or on pronsole? I would say the `readline` module is more critical on the latter, so a successful run on pronsole would indicate they are truly interchangeable?

DivingDuck, 2023-03-30:
Hi @rockstorm101, good point. I test it only on pronterface. I just start a needed little test print on pronsole and run monitor too.

DivingDuck, 2023-03-30:
The print runs fine with pronsole and activated monitor too. Guess, I can send an update later this evening.

commit ea5c8caa (2023-04-03): Change of module pyreadline for pyreadline3 (#1335)

Fixes #1332
closed 2023-04-03 by rockstorm101