kliment/Printrun #1340: Pronterface won't launch, AttributeError: 'NoneType' object has no attribute 'write'

naeim77, 2023-04-08:
AttributeError: 'NoneType' object has no attribute 'write'

I open the pronterface and I get this error

How can I fix it?

rockstorm101, 2023-04-08:
Please share your operating system, Printrun versión, where did you download it? Python version? How are you launching it?

volconst, 2023-04-08:
Try to find/enable log file
https://github.com/kliment/Printrun/issues/1295#issuecomment-1377125806

rockstorm101, 2023-05-11:
Hi @naeim77, did you manage to solve your issue? Did the suggestion help in anyway?

DivingDuck, 2024-10-17:
@rockstorm101,
This issue should be solved with Pronterface 2.1.0. Guess, we can close this issue.

rockstorm101, 2024-10-20:
> @rockstorm101, This issue should be solved with Pronterface 2.1.0. Guess, we can close this issue.

Agreed, this is clearly gone stale. Thanks for catching up.

closed 2024-10-20 by rockstorm101

FredBello, 2024-10-29:
My friends, in my case, not even the current one worked
Version: printrun-2.2.0_windows_x64_py3.10
Error message:
"Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
  File "printrun\pronterface.py", line 2596, in __init__
  File "printrun\pronterface.py", line 171, in __init__
  File "printrun\pronsole.py", line 778, in parse_cmdline
  File "printrun\pronterface.py", line 1161, in process_cmdline_arguments
  File "printrun\pronsole.py", line 761, in process_cmdline_arguments
  File "printrun\pronsole.py", line 656, in load_default_rc
  File "printrun\pronsole.py", line 627, in load_rc
  File "cmd.py", line 211, in onecmd
  File "printrun\pronsole.py", line 403, in default
  File "cmd.py", line 236, in default
AttributeError: 'NoneType' object has no attribute 'write'   "

I don't know what to do, could you give me some suggestions?
Thanks

DivingDuck, 2024-10-30:
For reference: #1462