kliment/Printrun #1341: Issue with firmware 3.12.2: Line Number is not Last Line Number+1

ssuchter, 2023-04-14:
I've been using Printrun (from source) for a couple years now on my Prusa MK3. A couple days ago I upgraded my firmware to 3.12.2-5713 and can no longer print. The same gcode files that worked last week, no longer work.

When I attempt to print anything, I get spammed with thousands of messages like this:

Resend: 8467
Error:Line Number is not Last Line Number+1, Last Line: 8466
ttyACM0 27° 97.6%> Error:Line Number is not Last Line Number+1, Last Line: 8466

Resend: 8467
Error:Line Number is not Last Line Number+1, Last Line: 8466
ttyACM0 27° 97.6%> Error:Line Number is not Last Line Number+1, Last Line: 8466

Resend: 8467
Error:Line Number is not Last Line Number+1, Last Line: 8466
ttyACM0 27° 97.6%> Error:Line Number is not Last Line Number+1, Last Line: 8466

Resend: 8467
Error:Line Number is not Last Line Number+1, Last Line: 8466
ttyACM0 27° 97.6%> Error:Line Number is not Last Line Number+1, Last Line: 8466

I'm utilizing the source as of f652640c0342d737a2315136ed8be0ef03cb9303.

Has anyone else been able to print via pronsole to a printer running firmware 3.12.2?

DivingDuck, 2023-04-14:
My first idea would be to ask Prusa if there is something wrong with the new firmware, especially if this wasn't the case with the old firmware and this was the only thing that had change.
I've seen some posts in the prusa forum on problems with this firmware, not the problem you had, but maybe you found an additional bug in the new firmware.

I don't have a Prusa printer but I did some tests with Pronsole last week based on the new version 2.0.0 of Pronterface in preparation of a modul change for a future version and didn't seen any flaws on my printer in processing test prints.

What version of pronterface / pronsole is installed on your computer?

ssuchter, 2023-04-14:
Here's an interesting datapoint too: I set up a Octoprint instance on the same server I was running Pronsole, connected via the exact same USB, running at the same user. The exact same gcode file, when sent via Octoprint, works fine.

So it has to be something about the manner in which the printer is interfacing with Pronsole.

I will absolutely go and report the issue in that thread, thanks.

CoderNtraining, 2023-04-18:
I'm having the same issue you're mentioning @ssuchter.  I'd be interested to know what you find out.

volconst, 2023-04-18:
@ssuchter can you enable debug communication checkbox from the menu and upload the log. It will be noisy.

> Resend: 8467
> Error:Line Number is not Last Line Number+1, Last Line: 8466

Not looking at the code, but 8467 is 8466 +1, maybe we never supported resending.

luckyNui, 2023-05-11:
Hello, I have the same issue where I encounter the same error. I am using the Python Printcore library. I have a Python script that is supposed to read a G-code file and then start a print using the .startprint() function, just like the example on the website.

However, even with a basic G-code script for movement (e.g., G1 X10 F1000...), I still receive the Line Number error.

Interestingly, if I run the same commands using the .send() function (e.g., .send("G1 X10 F1000")...), I do not encounter the error.

I am using a Prusa i3 MK3S+ with firmware version 3.12.2.

I have come across some suggestions on other forums that installing an older firmware version might solve the issue.

Do you have any ideas on how to fix this? I plan to try installing the older firmware later this week.

digitalcraft, 2023-05-12:
It's good to see this has so much activity on a highly repeatable issue.
++active community

volconst, 2023-07-18:
The text "Line Number is not Last Line Number+1, Last Line:" is coming from Prusa firmware.
https://github.com/prusa3d/Prusa-Firmware/blob/5295bfa0402c89aa8a45f8fc260b4135d542f3a6/Firmware/cmdqueue.cpp#L420
```
  SERIAL_ERRORRPGM(_n("Line Number is not Last Line Number+1, Last Line: "));////MSG_ERR_LINE_NO
```
Maybe pronterface does not reset correctly line numbers when starting a second print.

volconst, 2023-07-18:
> Do you have any ideas on how to fix this? I plan to try installing the older firmware later this week.

Try disconnect/connect before each print or exit the process/script before print.

dMAC95, 2023-08-15:
I fixed this by sending "M110 N-1" between prints in my printcore file.

dMAC95, 2023-08-15:
I'll create a pull request with the fix

dMAC95, 2023-08-15:
> I fixed this by sending "M110 N-1" between prints in my printcore file.

It seems printcore already tries to do this, but it sends it as "N-1 M110" instead of "M110 N-1" and the 3.12 firmware no longer supports the backwards notation of that command.

rockstorm101, 2023-08-20:
Fixed by commit 0f71ff3.

closed 2023-08-20 by rockstorm101