kliment/Printrun #1344: Unhandled exception in script

Shuda12, 2023-04-21:
I was trying to launch to the pronterface exe. 
I am getting this error
I am using windows 11 and the newest Python. 

 Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
  File "printrun\pronterface.py", line 2583, in __init__
  File "printrun\pronterface.py", line 170, in __init__
  File "printrun\pronsole.py", line 786, in parse_cmdline
  File "printrun\pronterface.py", line 1157, in process_cmdline_arguments
  File "printrun\pronsole.py", line 769, in process_cmdline_arguments
  File "printrun\pronsole.py", line 664, in load_default_rc
  File "printrun\pronsole.py", line 635, in load_rc
  File "cmd.py", line 211, in onecmd
  File "printrun\pronsole.py", line 412, in default
  File "cmd.py", line 236, in default
AttributeError: 'NoneType' object has no attribute 'write'

I downloaded the files through the windows powershell command.

DivingDuck, 2023-04-21:
Hi,
What version of Pronterface is running? Do you run Pronterface from source?

Shuda12, 2023-04-21:
The lastest release and through the pronterface.exe

DivingDuck, 2023-04-21:
can you pls. tell me the version number shown via -->Help -->About Printrun?

Shuda12, 2023-04-21:
It will not show me that as this happens the moments I try and run the exe. 
[image]

Shuda12, 2023-04-21:
I just downloaded 1.6.0 and it opens fine.

DivingDuck, 2023-04-21:
Sorry, I was interrupted....
unbelievable :)
I want to find out what happen with your installation. I can't get your error on my windows 10 computer. 
Had you download the newest version from this source: https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0 or from somewhere else?

Shuda12, 2023-04-21:
Correct it is from that release from github.

DivingDuck, 2023-04-21:
@Shuda12,
I searched a little bit in the code. I can't see a problem there. Looking in detail on your trace back report I think that there is maybe a problem with a corrupt configuration file called printrunconf.ini.
This file had caused in the past problems for some users. It mostly happen if a user switch between different Pronterface versions. What had helped was to delete this configuration file. Maybe you can try to do this as well.

The configuration file is possible  located on two places (you need to delete both before you start pronterface v2.0.0 again).
Change YOUR_USER_NAME to your user name:

**First place location** is the root of your user directory: c:\Users\YOUR_USER_NAME\\**printrunconf.ini**
--> delete the file **printrunconf.ini**
**Second place location** is directory c:\Users\YOUR_USER_NAME\AppData\Local\\**Printrun**
-->delete this directory **Printrun**

**Remark**: c:\Users\YOUR_USER_NAME\AppData\ is a hidden directory. (please see for further help if needed: https://support.microsoft.com/en-us/windows/view-hidden-files-and-folders-in-windows-97fbc472-c603-9d90-91d0-1166d1d9f4b5#WindowsVersion=Windows_11 )

It would be nice if you can make a backup of both, the mentioned printrunconf.ini and the directory printrun, before you delete them. Make a zip file and attach the zip file here. In case there is a file called Printrun.log available, please add it to the zip file too.  It should be located at c:\Users\YOUR_USER_NAME\\**Printrun.log**

Let me know whether this solve the problem or not.

DivingDuck, 2024-01-09:
@Shuda12 , 
haven't got any further feedback regarding your issue. In the meantime we published a newer version. You can find the actual version of Pronterface here: https://github.com/kliment/Printrun/releases

I will close this issue as it belongs to an older version of Pronterface. Feel free to open a new issue in case the issue is still existing and/or you have further information running the actual version.

Best regards, DD

closed 2024-01-09 by DivingDuck