kliment/Printrun #1345: unhandled exception

sujdovicluka, 2023-04-23:
When I run pronterface/ printrun, error occures:

I have no clue on how to solve this but this is what console says:

Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
  File "printrun\pronterface.py", line 2581, in __init__
wx._core.wxAssertionError: C++ assertion ""lang != wxLANGUAGE_UNKNOWN"" failed at ..\..\src\common\intl.cpp(425) in wxLocale::Init(): Initializing unknown locale doesn't make sense, did you mean to use wxLANGUAGE_DEFAULT perhaps?

[image]

Can you help me with this? /- my email is > sujdovic21@gmail.com
Thanks.

DivingDuck, 2023-04-23:
Hi @sujdovicluka 
Can you tell me what operating system are you running Pronterface and what version is used?

sujdovicluka, 2023-04-23:
Windows 10 Pro, 20H2, 2021. , Windows Feature Experience Pack 120.2212.2020.0,

DivingDuck, 2023-04-23:
What version of Pronterface did you try to run?
What is the language setting of your computer?

sujdovicluka, 2023-04-23:
https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0 this is the latest, I have no access to the older version.

Language - English?

DivingDuck, 2023-04-23:
Thanks for this additional information. 
I'm running as well Windows 10 Pro, but with latest version 22H2.
I'm just fishing and try to understand what is different to my system as I can't force this error here.

Maybe a additional question, is English meant en-US or similar or something else?

Background: 
We had an user a couple weeks ago on a macOS who was using a language setting like en-AT, what is not a supported combination for pronterface, but possible to set up. He needed to switch to a common combination like de-AT or en-GB, en-US etc. 
Peoples choose those kind of settings usually if they have a different keyboard layout like for example a French system language with an English keyboard layout.

neofelis2X, 2023-04-23:
It may actually be a similar but not the same problem.

In this case it could be that wx.Local doesn't recognise a language it supports. Unfortunately, there is no safeguard to check if _GetSystemLanguage_ was actually successful.

wx.LANGUAGE_UNKNOWN | Unknown language, **returned if wx.Locale.GetSystemLanguage fails.**
Source: https://docs.wxpython.org/wx.Language.enumeration.html#wx-language

rockstorm101, 2023-05-10:
Same issue as #1348. Reproduced. Not platform specific.

PR #1351 (2023-05-13): p/pronterface: Fix unknown wx locale
Function `wx.Locale.GetSystemLanguage` returns `wx.LANGUAGE_UNKNOWN` when unable to identify the system's language. And `wx.Locale` constructor does not allow `wx.LANGUAGE_UNKNOWN` as an argument.

Fixes #1345
Fixes #1348

CC @DivingDuck @neofelis2X could you give this a try and confirm it works?
commit 9c165406 (2023-05-13): p/pronterface: Fix unknown wx locale (#1351)

Function `wx.Locale.GetSystemLanguage` returns `wx.LANGUAGE_UNKNOWN`
when unable to identify the system's language. And `wx.Locale`
constructor does not allow `wx.LANGUAGE_UNKNOWN` as an argument.

Fixes #1345
Fixes #1348
closed 2023-05-29 by rockstorm101