kliment/Printrun #1348: Failed to execute script 'pronterface' due to unhandled exception: C++ assertion

379820, 2023-04-27:
/Applications/pronterface.app/Contents/MacOS/pronterface 
```
Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
  File "printrun/pronterface.py", line 2581, in __init__
wx._core.wxAssertionError: C++ assertion ""lang != wxLANGUAGE_UNKNOWN"" failed at /Users/robind/projects/bb2/dist-osx-py310/build/ext/wxWidgets/src/common/intl.cpp(425) in Init(): Initializing unknown locale doesn't make sense, did you mean to use wxLANGUAGE_DEFAULT perhaps?
[41290] Failed to execute script 'pronterface' due to unhandled exception: C++ assertion ""lang != wxLANGUAGE_UNKNOWN"" failed at /Users/robind/projects/bb2/dist-osx-py310/build/ext/wxWidgets/src/common/intl.cpp(425) in Init(): Initializing unknown locale doesn't make sense, did you mean to use wxLANGUAGE_DEFAULT perhaps?
[41290] Traceback:
Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
  File "printrun/pronterface.py", line 2581, in __init__
wx._core.wxAssertionError: C++ assertion ""lang != wxLANGUAGE_UNKNOWN"" failed at /Users/robind/projects/bb2/dist-osx-py310/build/ext/wxWidgets/src/common/intl.cpp(425) in Init(): Initializing unknown locale doesn't make sense, did you mean to use wxLANGUAGE_DEFAULT perhaps?
```

[0] % locale 
```
LANG="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_CTYPE="UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_ALL=
```

uname -a
`Darwin MacBook-Pro 22.4.0 Darwin Kernel Version 22.4.0: Mon Mar  6 21:00:17 PST 2023; root:xnu-8796.101.5~3/RELEASE_X86_64 x86_64`

ZoyaZ126, 2023-05-04:
Hi, I had the same problem until I changed my system language to English…

DivingDuck, 2023-05-04:
Interesting. What was your language setup before?

ZoyaZ126, 2023-05-04:
Chinese

DivingDuck, 2023-05-04:
Thanks. Good to know. Maybe generating/making a translation file could  help.

rockstorm101, 2023-05-10:
I'm able to reproduce this on my Linux box so I've removed the 'macOS' flag.

dymbox, 2023-05-12:
Same here for macOS 11.7.6, added US-English locale to .zshrc with no success. My current locale is:

LANG="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_CTYPE="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_ALL="en_US.UTF-8"

Sure I'm running the macOS 11 version, but tried both 10.15 & 12 too, same result.

BTW, the Printrun 2.0.0rc8 runs just fine

PR #1351 (2023-05-13): p/pronterface: Fix unknown wx locale
Function `wx.Locale.GetSystemLanguage` returns `wx.LANGUAGE_UNKNOWN` when unable to identify the system's language. And `wx.Locale` constructor does not allow `wx.LANGUAGE_UNKNOWN` as an argument.

Fixes #1345
Fixes #1348

CC @DivingDuck @neofelis2X could you give this a try and confirm it works?
commit 9c165406 (2023-05-13): p/pronterface: Fix unknown wx locale (#1351)

Function `wx.Locale.GetSystemLanguage` returns `wx.LANGUAGE_UNKNOWN`
when unable to identify the system's language. And `wx.Locale`
constructor does not allow `wx.LANGUAGE_UNKNOWN` as an argument.

Fixes #1345
Fixes #1348
closed 2023-05-29 by rockstorm101