kliment/Printrun #1349: Mac OS Ventura not compatible

mallerb, 2023-04-29:


neofelis2X, 2023-04-29:
Hi there. I just tested the Printrun release version 2.0.0 (download printrun-2.0.0_**macos-12**_x64_py3.10.zip) on macOS Ventura 13.3.1 (M2 chip) and it runs just fine. Takes a little long to open (~15s), but once it did, everything seems to work normally.

I may quote this from the Readme, a common problem on macOS:
> Note for OSX users: if OSX tells you "pronterface.app" cannot be opened because the developer cannot be verified., you don't need to redownload it. Instead, you need to allow OSX to run the unsigned app. To do this, right click the application in Finder and select Open. Then click Open in the popup window that appears. You only need to do this once.

So if there was a different problem on the Mac, you would have to be a bit more specific.

DivingDuck, 2024-10-18:
I close this issue as there was no feedback

closed 2024-10-18 by DivingDuck