kliment/Printrun #1350: MacOS Ventura MacBook Pro M2 printer won't launch after custom button added

traceyflanders, 2023-05-02:
Hello,

I get an error when trying to launch Pronterface after it crashed added a custom button. It did work for a few days until I added a custom button. Is there a user settings file I can delete to remove the customization? I couldn't find anything in my home directory that was related.

This is what I get when I launch it via terminal.

### OUTPUT ###
user@AlphaMusk-M2-MacBook-Pro ~ % /Applications/pronterface.app/Contents/MacOS/pronterface
2023-05-01 17:53:09.924 pronterface[1386:21749] ApplePersistenceIgnoreState: Existing state will not be touched. New state will be written to /var/folders/t5/htk9tx7n07bck2nt_840m_4c0000gn/T/pronterface.savedState
Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
  File "printrun/pronterface.py", line 2582, in __init__
  File "printrun/pronterface.py", line 219, in __init__
  File "printrun/pronterface.py", line 293, in reload_ui
  File "printrun/gui/__init__.py", line 295, in createGui
  File "printrun/pronterface.py", line 2155, in cbuttons_reload
wx._core.wxAssertionError: C++ assertion "CheckExpectedParentIs(w, m_containingWindow)" failed at /Users/robind/projects/bb2/dist-osx-py310/build/ext/wxWidgets/src/common/sizer.cpp(850) in DoInsert(): Windows managed by the sizer associated with the given window must have this window as parent, otherwise they will not be repositioned correctly.

Please use the window wxPanel@0x7fe1b2f97150 ("panel") with which this sizer is associated, as the parent when creating the window wxStaticText@0x7fe1b2f9df90 ("staticText") managed by it.
[1386] Failed to execute script 'pronterface' due to unhandled exception: C++ assertion "CheckExpectedParentIs(w, m_containingWindow)" failed at /Users/robind/projects/bb2/dist-osx-py310/build/ext/wxWidgets/src/common/sizer.cpp(850) in DoInsert(): Windows managed by the sizer associated with the given window must have this window as parent, otherwise they will not be repositioned correctly.

Please use the window wxPanel@0x7fe1b2f97150 ("panel") with which this sizer is associated, as the parent when creating the window wxStaticText@0x7fe1b2f9df90 ("staticText") managed by it.
[1386] Traceback:
Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
  File "printrun/pronterface.py", line 2582, in __init__
  File "printrun/pronterface.py", line 219, in __init__
  File "printrun/pronterface.py", line 293, in reload_ui
  File "printrun/gui/__init__.py", line 295, in createGui
  File "printrun/pronterface.py", line 2155, in cbuttons_reload
wx._core.wxAssertionError: C++ assertion "CheckExpectedParentIs(w, m_containingWindow)" failed at /Users/robind/projects/bb2/dist-osx-py310/build/ext/wxWidgets/src/common/sizer.cpp(850) in DoInsert(): Windows managed by the sizer associated with the given window must have this window as parent, otherwise they will not be repositioned correctly.

Please use the window wxPanel@0x7fe1b2f97150 ("panel") with which this sizer is associated, as the parent when creating the window wxStaticText@0x7fe1b2f9df90 ("staticText") managed by it.
### END OUTPUT ###

My Mac Info
M2-MacBook-Pro ~ % systemstats

Summary
=======
           System Version:	22E261
           Hardware Model:	Mac14,6
                 Board ID:	06000000
    Installed Memory Size:	32GB (34359738368)
                Page Size:	16384
         Root Volume Type:	SSD:1,HDD:0,FUSION:0,FDE:1,RAID:0,TINYSSD:0,IN:1,EX:0,INEX:0,APFS
         Root Volume Size:	926GB (994662584320)
                 SSD Name:	APPLE SSD AP1024Z
             SSD Revision:	874.100.
     Discrete GPU Present:	0
               Total Time:	555:48:47
          Dashboard Start:	2023-04-08 21:16:37
            Dashboard End:	2023-05-02 01:05:13

DivingDuck, 2023-05-02:
Hi, can you please attach your log- and config file as zip file? Search for printrunconf.ini and Printrun.log. 
The location for the log file can be found here:

[image]
[image]

For the config file please search for it as I don't know where the locations are for macOS. In Windows they have two locations (you need to delete both), so better search for it.
Make a backup of the file first and then delete it. Pronterface will create an initial new one as soon as you launch it again.

neofelis2X, 2023-05-02:
The path is : /Users/**Your_Username**/Library/Application Support/Printrun/
_pronsolerc_ can be edited in any Texteditor. 
As for the buttons, the index of the buttons in the settings file (button 0, button 1, button 2, etc...) must have an ascending sequence. If one index is missing (0, 1, _, 3) the loading of the buttons will fail. The _order_ of the indices doesn't matter tho.

DivingDuck, 2023-05-02:
Yea, I still forgot that macOS and Linux use a different file name for the config file...

DivingDuck, 2023-05-04:
@traceyflanders, 
had this solve your problem?

DivingDuck, 2025-02-15:
I will close the issue because of inactivity and no further response since long time.

closed 2025-02-15 by DivingDuck