kliment/Printrun #1354: MacOS Ventura MacBook Pro : Can't install Printrun on

codeZ-1627, 2023-05-11:
[image]

rockstorm101, 2023-05-12:
This seems to be a known issue on wxPython. Ref https://github.com/wxWidgets/Phoenix/issues/2296. Seems like Python 3.11 is not yet supported. Maybe try installing with Python 3.10?

DivingDuck, 2023-05-12:
Hi,
Had you recognize that this version is not the actual version?

rockstorm101, 2023-05-12:
On a side note, the Printrun version on PyPI. Let's see if we can update that on a (hopefully coming soon) 2.0.1 version.

codeZ-1627, 2023-05-12:
Thx for helping, I able to install it after installing attrdict3, wxPython, then Printrun.
Also excited for the hopefully coming soon version :)

closed 2023-05-12 by codeZ-1627