kliment/Printrun #1357: Deprecation of disutils module

rockstorm101, 2023-05-15:
As per [PEP 632][1] the `distutils` module is now officially deprecated and will be removed in Python 3.12. As far as I can see we only import `distutils` on the `setup_win.py` file. Do we even use this file at the moment? CC @DivingDuck 

[1]: https://peps.python.org/pep-0632

DivingDuck, 2023-08-13:
Hi @rockstorm101,

I had totally overseen this issue. Sorry.
We don't use setup_win.py file. Guess, this was a relic from from the time where Pronterface was build with python 2.7. This file can go in retirement :). 

For windows builds we use the file setup.py too. Looking on that file I recognize there should be some adjustments made as well (e.g. supported Python versions: 3.6 isn't supported for actual wxPython, add py 3.9 and 3.10).

Best regards, DD

rockstorm101, 2023-08-14:
Thanks I'll proceed with the "retiring" then :P

> Looking on that file I recognize there should be some adjustments made as well (e.g. supported Python versions: 3.6 isn't supported for actual wxPython, add py 3.9 and 3.10).

Good catch. I'll do that as well.

commit 067e7f24 (2023-08-16): Remove setup_win.py and fix supported Pythons (#1375)

Also dropped the Python 3.6 wheel build for Linux. Not supported by wxPython anyway but also it triggers a failure since setup.py now requires Python >= 3.7.

Fixes #1357
closed 2023-08-16 by rockstorm101