kliment/Printrun #1358: Can not get Printrun (Pronterface) to run on the latest Umbuntu 21.1

Arnie3d, 2023-05-19:
Hhi eveyone this is an older problem but is it over using Printrun on Mint?
I did the work around ./pronterface.py in the dir of printrun 
using a System:
  Kernel: 5.15.0-72-generic x86_64 bits: 64 compiler: gcc v: 11.3.0 Desktop: Cinnamon 5.6.8
    tk: GTK 3.24.33 wm: muffin dm: LightDM Distro: Linux Mint 21.1 Vera base: Ubuntu 22.04 jammy
Machine:
  Type: Laptop System: TOSHIBA product: SATELLITE L775-18Q v: PSK3WE-05X00UDU
    serial: <superuser required> Chassis: type: 9 serial: <superuser required>
  Mobo: Intel model: Oneonta Falls serial: <superuser required> BIOS: American Megatrends
    v: 1.70 date: 08/22/2011
CPU:
  Info: dual core model: Intel Core i5-2450M bits: 64 type: MT MCP arch: Sandy Bridge rev: 7
    cache: L1: 128 KiB L2: 512 KiB L3: 3 MiB
  Speed (MHz): avg: 798 min/max: 800/3100 cores: 1: 798 2: 798 3: 798 4: 798 bogomips: 19956
  Flags: avx ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3
Graphics:
  Device-1: NVIDIA GF108M [GeForce GT 525M] vendor: Toshiba driver: nouveau v: kernel pcie:
    speed: 2.5 GT/s lanes: 16 ports: active: LVDS-1 empty: HDMI-A-1,VGA-1 bus-ID: 01:00.0
    chip-ID: 10de:0dec
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: modesetting unloaded: fbdev,vesa
    gpu: nouveau display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1600x900 s-dpi: 96
Repos:
  Packages: 2420 apt: 2408 flatpak: 12
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com vera main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
Info:
  Processes: 239 Uptime: 2h 42m Memory: 15.61 GiB used: 1.94 GiB (12.4%) Init: systemd v: 249
  runlevel: 5 Compilers: gcc: 11.3.0 alt: 11/12 Client: Unknown python3.10 client inxi: 3.3.13
WORK AROUND worked but what does this all mean 
arnie@arnie-SATELLITE-L775-18Q:~/Printrun$ ./pronterface.py
WARNING:root:Memory-efficient GCoder implementation unavailable: cannot import name 'gcoder_line' from 'printrun' (/home/arnie/Printrun/printrun/__init__.py)
WARNING:root:RPC server bound on non-default port 7984

(pronterface.py:14067): Gtk-CRITICAL **: 22:55:17.347: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:14067): Gtk-CRITICAL **: 22:55:17.347: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:14067): Gtk-WARNING **: 22:55:17.357: Negative content width -1 (allocation 1, extents 1x1) while allocating gadget (node scrolledwindow, owner GtkScrolledWindow)

(pronterface.py:14067): Gtk-WARNING **: 22:55:17.357: Negative content width -17 (allocation 1, extents 9x9) while allocating gadget (node entry, owner GtkEntry)

(pronterface.py:14067): Gtk-WARNING **: 22:55:17.358: Negative content width -33 (allocation 1, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:14067): Gtk-WARNING **: 22:55:17.358: Negative content width -33 (allocation 1, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:14067): Gtk-WARNING **: 22:55:17.360: Negative content width -13 (allocation 5, extents 9x9) while allocating gadget (node entry, owner GtkEntry)

(pronterface.py:14067): Gtk-WARNING **: 22:55:17.469: Negative content width -13 (allocation 5, extents 9x9) while allocating gadget (node entry, owner GtkEntry)
arnie@arnie-SATELLITE-L775-18Q:~/Printrun$

rockstorm101, 2023-05-19:
Hi, yes, on Ubuntu 22.04 (and it's derivatives?) it works with the workaround as you mention. Those warnings are (no surprise here) just warnings and they don't affect the functionality I believe.

Arnie3d, 2023-05-19:
Thanks Rocstorm...Then it looks like were doomed :) 
Starting Pronterface with the work around every time i need it,  sure sucks and is not what i had in mind with Mint updates that all my 3D software wouldn't run on it anymore.
Does anyone know if if this is because of Python3 .10 or WX not working with Mint

rockstorm101, 2023-05-19:
> Does anyone know if if this is because of Python3 .10 or WX not working with Mint

It is due to an issue with Python 3.10. The Ubuntu packagers included the fix on Ubuntu 22.10. So any versions prior to that will have to either use the workaround or run the latest Printrun from source. Sadly, Linux Mint 21 is based on Ubuntu 22.04. Maybe ask the Linux Mint packagers (or the Ubuntu ones) to back-port the fix? Other option could be to try to install the package(s) from Ubuntu 22.10? (not recommend if you don't really know what you are doing, might break the system) Probably the safest option is to just run from source to be honest.

Arnie3d, 2023-05-19:
I will ask the crew of Ubuntu to take a look what they can do ...thank you very much for the info.

closed 2023-05-19 by Arnie3d