kliment/Printrun #1360: Projector: TypeError on loading SVG

rockstorm101, 2023-05-29:
1. Launch Pronterface
2. Tools > Projector
3. Load
4. Choose a random SVG file (one I produced with Inkscape not related to 3D printing)

```
Traceback (most recent call last):
  File "/home/rock/Projects/printrun/rockintrun/printrun/projectlayer.py", line 595, in load_file
    layers = self.parse_svg(name)
             ^^^^^^^^^^^^^^^^^^^^
  File "/home/rock/Projects/printrun/rockintrun/printrun/projectlayer.py", line 507, in parse_svg
    z = float(i.get('{http://slic3r.org/namespaces/slic3r}z'))
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: float() argument must be a string or a real number, not 'NoneType'
```
I guess we should check here for existence of that particular element before doing the type conversion? I'm not overly familiar with the Projector feature to be honest.

neofelis2X, 2023-05-29:
The thing is that parse_svg() only checks if there is a 'metadata' tag and treats those like skeinforge-files. _Everything_ else is declared as a slic3r file. So we would need another check that identifies actual slic3r files and sorts out anything else.

DivingDuck, 2023-05-30:
These are sliced svg files of an model. I used slic3er for generating this kind of file. Here is an example:
[Drehmeiselbrett v3-exp.zip](https://github.com/kliment/Printrun/files/11601433/Drehmeiselbrett.v3-exp.zip)

closed 2023-11-08 by rockstorm101