kliment/Printrun #1362: Pronterface broken in Python 3.9 or older

rockstorm101, 2023-06-15:
I just realized that on #1331 we introduced code that only runs under Python 3.10 or newer, meaning that it throws an "invalid syntax error" when run under Python 3.9. 

https://github.com/kliment/Printrun/blob/ceeb651ca9fa4c20757ba37128b552dde4662952/printrun/gui/widgets.py#L25

So, options:
 1. Change the code to be compatible with older versions? or
 2. Acknowledge and accept that the next Printrun 2.1 release will be only for Python >= 3.10?

Just a note on Debian/Ubuntu ecosystem, Ubuntu 20.04 (which ships Python 3.8) is supported until 2025.

kliment, 2023-06-15:
If that's the only place it's used, I think reworking it to be a bunch of `elif`s is reasonable.

kliment, 2023-06-15:
The other way to reformat it would be a bunch of dictionaries. It might actually be a better use case for readability. We'd then have one dictionary per platform.

neofelis2X, 2023-06-15:
Damn, that’s true. But I have already rewritten the whole function with a dictionary. I realised that it’s not a very elegant approach.
It’s in one of my branches, but I can also push it as a separate PR if you prefer that.

> We'd then have one dictionary per platform.

That’s almost exactly what I did in the new version. 😅

rockstorm101, 2023-06-16:
> It’s in one of my branches, but I can also push it as a separate PR if you prefer that.

Hi @neofelis2X I'm not sure I understand 100% what you mean here sorry. If you were going to submit the rewritten function as part of a bigger PR, then that's OK with me, it can wait. If you weren't going to submit it but happen to have that function already rewritten, then yes please send a PR with the "fix" :)