kliment/Printrun #1363: Issue with 2010 MacBook and PronterFace

Really01101001, 2023-07-08:
Before someone references a post that doesn't help was closed that references another post that doesn't help that references the previous post that didn't help please take a bit to help me actually solve the problem I'm having.

Using a mid 2010 MacBook pro, when I opened PronterFace there was a "resource" already entered in the Port window and only two choices in the drop down 
/dev/tty.motogplay2021-Secure or /Dev/cu.motogplay2021-Secure

I'm not very computer savvy but it seems totally wrong that my phone or something related to my phone is reference/a directory or whatever there. I thought it could have to do with the fact that I downloaded printrun to my phone then transferred to SD and then to the laptop. 

I uninstalled and then downloaded directly with/to the laptop. When I opened it the same motogplay was there.
I have looked at multiple forums for answers including apple forum and of course GitHub. Perhaps this isn't an issue with PronterFace perse but help is appreciated and please don't close this until the issue is actually solved.

neofelis2X, 2023-07-08:
Hello @Really01101001 !
Pronterface lists **all** ports it can find in the drop down menu. For example, on my system, when Bluetooth is enabled, Pronterface shows my Bluetooth headset here in the list of ports. (MAJORIV)
[image]

Of course, I cannot connect to the headset. But as soon as I connect my 3D printer via USB and refresh the list (with the 'Port' button), I can select the printer from the list.
So if your phone has been connected to your computer in the past, it is normal for Pronterface to show this port in the list.

Which version of **macOS** are which version of **Printrun** are you using?
Are you trying to connect a printer to Pronterface?

Cheers

Really01101001, 2023-07-09:
I found the box that needed to be checked to pick up the ports in PronterFace. I'll post a a little later screenshot/procedure for this for anyone having the same issue in the future.

rockstorm101, 2023-07-18:
Am I right in thinking this is not a Printrun bug?

Really01101001, 2023-07-18:
No not a bug. I'm not extremely computer savvy and new to GitHub. I didn't know to post "Not a bug" label. I did post an update for the issue/resolution I found a.s.a.p. hope that helped.

Really01101001, 2023-07-18:
Have used PronterFace successfully since.

rockstorm101, 2023-07-19:
> Have used PronterFace successfully since.

Glad to hear that. I'll close this issue then. Happy printing!

closed 2023-07-19 by rockstorm101