kliment/Printrun #1365: Strange serial connection issue

pedjas, 2023-07-17:
After latest update of Pronterface (2.0.1) and firmware in my Prusa MK3S (3.12.2-5713) I get quite strange behavior when printing from Pronterface.

First print run goes OK, but if i repeat print, Pronterface behaves like it prints (print simulation shows progress, but verz fast, adn it does not print, only displays this error on each command:

```
Resend: 244678
Error:Line Number is not Last Line Number+1, Last Line: 244677
Error:Line Number is not Last Line Number+1, Last Line: 244677

```
I have to close serial connection, reconnect and only then I am able to print. This happens each time.

I noticed that when I run predefined macro, that sets bed and nozzle temperatures, that macro works fine even when printing does not.

I did update of Pronterface and printer firmware almost at the same time, so I cannot tell which one caused this issue.

rockstorm101, 2023-07-18:
Duplicate of #1341 ?

volconst, 2023-07-18:
> Duplicate of #1341 ?

yes

closed 2023-08-20 by rockstorm101