kliment/Printrun #1368: Install printrun on a i386 (x32) architecture

Fedetecnico07, 2023-07-23:
Install Printrun on a x386 (32-bit) computer

Good afternoon. I wanted to ask you if it is possible to install Printrun, pronterface, etc. on an old computer, with a 32-bit x386 architecture.

I don't care if it's windows or linux, I've tried to install windows XP to optimize resources (honestly I didn't try to install windows 10 on the PC).

The computer has a 40GB hard drive, 256 ram and an AMD semptron processor (x86, i386).

The idea is to only use pronterface on the computer. I am unable to install it. I have tried everything, I tried 5 linux distributions, I tried windows, etc.
The problem is because it is an architecture that is not being supported.

In a nutshell:
Printrun needs python 3.9, but python 3.9 requires 64 bit architecture.
I have not been able to successfully install python and consequently pronterface.

I tried to use old versions of pronterface (V1.6), I searched for many kinds of solutions but I couldn't find how to effectively install pronterface (printrun) on an x32 bit operating system.

I hope you can guide me or confirm if this is possible since I can't find concrete answers to this problem.

Thank you

rockstorm101, 2023-07-24:
Hi @Fedetecnico07, architecture won't be a problem. For instance, you can run pure Debian on 32-bit architectures. And Printrun will be available from official repositories there (which means installing it is as simple as `sudo apt install printrun`). However, looking at those specs the 256 MB of RAM might be a limiting factor.

> [...], but python 3.9 requires 64 bit architecture.

Don't know where you got that but I don't think that's true. For instance, Debian ships even Python 3.12 for i386 architectures [^1].

[^1]: https://packages.debian.org/sid/i386/python3.12/download

Fedetecnico07, 2023-07-24:
Thanks RockStorm. I have also given up.

I don't know if it's worth telling you because of all the advice I've found on the net, they have suggested installing different distributions. In the meantime, trying to install pronterface in distributions that weren't complete, I ran into other problems.

Finally, thanks for your patience. **If you find it necessary to delete this topic, just do so**. Believe me I will keep it in mind for future projects.

Thanks for everything.

rockstorm101, 2023-07-25:
> I have also given up. [...] they have suggested installing different distributions.

Well that sounds a lot like the Linux ecosystem. There's hundreds of options out there, which does not make it easy for beginners. However, choosing a Linux distribution is outside the scope of this forum. Nonetheless, I can say that Printrun is packaged and shipped in several official repositories. Installing Printrun on any Linux distribution derived from Debian/Ubuntu, Fedora or Archlinux should work as [detailed in the instructions][1].

I'll close the issue now since the architecture question is answered. Don't hesitate to open a new one if you face any issues. As Henry Ford said: “When everything seems to be going against you, remember that the airplane takes off against the wind, not with it.”.

[1]: https://github.com/kliment/Printrun#linux

closed 2023-07-25 by rockstorm101