kliment/Printrun #1369: Can't open pronterface

Chrisfbell, 2023-07-30:
I am trying to run Pronterface on Ubuntu, without success. Ubuntu version is 22.04 Jammy: Pronterface version 2.0.0 rc7: Python version3.10.6.
When I click on the Pronterface Icon in "activities" I see rotating arrows at the top of the screen, but it eventually times out.

rockstorm101, 2023-07-30:
Hi, this is likely a duplicate of #1225. Please have a look at the following two comments for potential workarounds. 

https://github.com/kliment/Printrun/pull/1224#issuecomment-1133590834
https://github.com/kliment/Printrun/pull/1224#issuecomment-1613648722

DivingDuck, 2024-10-18:
@rockstorm101,
should we close this issue?

rockstorm101, 2024-10-20:
> @rockstorm101, should we close this issue?

Agreed. No feedback whatsoever for quite a while.

closed 2024-10-20 by rockstorm101