kliment/Printrun #1372: Strange green lines on the PLATER window

JosRichters, 2023-08-11:
Hi there, I am new to Pronterface and I have installed it on two of my computers.
One is an Windows 11 HP Pavilion, on witch Pronterface seems to works fine. I use this one to just have a look at and learn.
The other is a Windows 10 Home Acer type A315-54. This computer has an USB connection with my Anycubic Mega Pro 3-D printer.
This is my problem: Pronterface is not showing the center window well.
See the image below; the center window shows the green lines.
[image]

Below is what is on my HP computer:
[image]

Can somebody tell me what is wrong here and what I can do to get Pronterface working as designed on this Acer computer (it is a laptop actually....)?

Thanks!
Jos.

DivingDuck, 2023-08-11:
Hi @JosRichters,
can you tell us what version of Pronterface you are running (Menu: -->Help -->About Printrun)? 
Looking to the first picture my first impression was that you had zoomed deeply, but I guess you had already check this. What happen if you hide the console window on the right side (click the [>] icon button or the left window with the [<] icon button?

Also check the version of your graphic card driver. Maybe there is a incompatibility with Pronterface we do not know about.

Looking to the second window, Pronterface try to slice a .stl file with sli3r. Seems slic3r isn't available or you didn't set the parameters correctly.

JosRichters, 2023-08-11:
Hi DivingDuck,
Please give me some time to get your comment sorted out.
Thanks for your prompt comment anyway.

Cheers,
Jos.

JosRichters, 2023-08-13:
Hi DivingDuck,
About Pronterface; I have installed the latest version: **Printrun 2.0.1** on all of my computers that I use to test and actual 3-D print on my Anycubic.
To give you complete information about the Acer:
[image]

When I hide the console window on my Acer I get the below situation:
[image]

Yesterday I spend the whole day on experimenting on various 3-D slicers, Youtube's and more. I also installed Slic3R on my computers and got the whole working. I also installed Pronterface with Slic3R on another laptop, a HP Probook that still runs on Windows 10 Pro. I loaded the 20mm test cube and connected my Anycubic to this laptop and started the printjob. All went well as you can see on the picture below. So Pronterface installed perfectly on this old computer.

[image]

DivingDuck, 2023-08-13:
Hi,
this graphic problem on your Acer looks like a hardware or a graphics driver problem - at least this is my best guess. Maybe it is worthwhile updating your windows 10 to actual version 22H2.

Glad to hear that your other installation is running well now.

Maybe a tip from my side regarding slicers. You can integrate other slicers with command line interface as well. For Superslicer you can use the same commands as for Slic3r. The easiest way to automate toggling between slicers is to use macros and custom buttons.

Here are two macros for Prusa- and Superslicer:
First create the macros (--Settings -->Macros -->New)
You need to change the path of each Slicer to your locations

[image]

[image]

```
macro SuperSlicer
  set slicecommandpath C:\\Users\\Armin\\SuperSlicer_2.4.58.5_win64_220924\\
  set slicecommand superslicer_console.exe $s --output $o
  set sliceoptscommand superslicer.exe --save SuperSlicer.ini --ignore-nonexistent-config
```
```
macro Slic3r
  set slicecommandpath c:\\Users\\Armin\\Slic3r-master.2020.08.24\\
  set slicecommand slic3r-console $s --output $o
  set sliceoptscommand slic3r --autosave slic3rconf.ini --ignore-nonexistent-config  
```
```
macro PrusaSlicer
  set slicecommandpath c:\\Users\\Armin\\PrusaSlicer-2.6.0+win64-202306191221_signed\\
  set slicecommand prusa-slicer-console $s --output $o -s
  set sliceoptscommand prusa-slicer.exe --save PrusaSlicer.ini --ignore-nonexistent-config
```
Remark: You need two backslashes for folder locations and for macro names upper- and lowercase matters.

Then create 3 custom custom buttons and use a command the macro name (click on the [+] button icon below the print bed) like:

[image]

```
Button Title: SuperSlicer
Command: SuperSlicer
Color:  #3485ff
```
```
Button Title: PrusaSlicer
Command: PrusaSlicer
Color: #ed6b21
```
```
Button Title: Slic3R
Command: Slic3r
Color: #9c9c14
```
Remark: Input of color is optional

JosRichters, 2023-08-13:
Thank for your comment DivingDuck!
I will try to upgrade my Acer OS to Windows 11.
I;ll let you know the results.

Concerning slicers,
Can I also setup for Cura?
And what would the parameters be if it is so?

Thanks.
Jos

JosRichters, 2023-08-13:
Pitty! No luck.
I upgraded the Acer to Windows 11 and started Pronterface.
The problem persist.....

I can use Pronterface on my olp HP Probook, so I will do thet when I need to do so, otherwise I will use Cura on the Acer.
Thanks for now.....

Cheers,
Jos.

DivingDuck, 2023-08-13:
This is bad. Had you check your graphic device driver? Is it an Intel so it is maybe worth to check on the intel device driver pages directly.
An other thing what came in mind is testing if set the visualization setting from 3D to 2D helps. I had an issue long time ago on an really old laptop that run out of shared memory. Easy to test: -->Settings -->Options -->Viewer -->Main visualization

Regarding Cura, no I haven't a solution for this slicer. It is way more complicated as you need to use CuraEngine for that and you need to to set up most things manually and gave up on the game changing run with every new release, For me it wasn't worth the time doing this. 

There was a guy in the past who wrote a command line wrapper but I think he stop the project as well some when in the past - at least I didn't saw any further updates of this project. Reference:
https://xyzdims.com/2021/03/24/3d-printing-cura-cli-wrapper-cura-slicer/
https://github.com/Spiritdude/Cura-CLI-Wrapper

JosRichters, 2023-08-13:
WOW!!! 2-D did the job!
[image]
But when I open the G-code Plater the green lines are still there as you can see.

Question: How do I do that? Checking the Intel device driver pages? And where do I have to look for? I have never did this before..
And I have another Acer laptop (leftover from a big project). I'll install Pronterface on that one as well, just to see if there is a physical error in the graphics board.  It seems odd to me that two identical laptops have the same malfunction in the hardware, but double check is better.

I will have a look at the references you suggested.

No problem with Cura.... I can use that standalone anyway...

Thanks again.
Jos.

DivingDuck, 2023-08-14:
Regarding Plater missing 2D view:
I think this behavior seams to be wrong. I create a new issue #1373 for this so that it won't be forgotten.

Graphic card driver:
Yes I think you should look at the device driver page from Intel. This is maybe a good starting point:
Driver and Support Assistant https://www.intel.com/content/www/us/en/support/articles/000005629/graphics.html

JosRichters, 2023-08-14:
Thank you very much for what you have done so far!

I'll wait for the update and I will happily test it when available.
Just let me know.
Jos.

feliks912, 2024-11-29:
Setting anti-aliasing to >0 fixed this issue on my Windows 10, RTX2070 machine.