kliment/Printrun #1382: Falha Nine Type

joserobertoalima, 2023-09-12:


joserobertoalima, 2023-09-12:
Falha para iniciar o Pronterface "nine type"

rockstorm101, 2023-09-13:
Hi @joserobertoalima, welcome to GitHub and to Printrun/Pronterface. To understand your issue and be able to help you we need:
 1. Problem/error description in English (this is an English-speaking forum)
 2. Operating system and version, your version of Printrun and how you installed it
 3. Error traces/debugging information

As a general advise for internet forums: the better the issue description, the easier to get help.

DivingDuck, 2023-10-07:
@rockstorm101, guess we can close this issue as there is no further feedback.

rockstorm101, 2023-10-09:
> Maybe this is an duplicate of issue #1340.
> Solution can be found here: [#1295 (comment)](https://github.com/kliment/Printrun/issues/1295#issuecomment-1377125806)

Well spotted @DivingDuck! I agree, please go ahead and close issues when you see fit

closed 2023-10-13 by rockstorm101