kliment/Printrun #1384: Can't open because Apple can't check malicious software and request an update

im-ellipsism, 2023-09-22:
Hello, I want to use Prontface for the first time because I changed the extruder of my printer, but it is not working because an alert opens when I try to run it, like in the attached image. It says that it cannot be opened because it cannot check if it contains malicious software, and asks to update the software. It can be fixed?. My OS is macOS Ventura 13.4.

Thank you so much.

[image]

DivingDuck, 2023-09-22:
You forgot to mention the version of Pronterface, Is this the actual Pronterface version? 
I guess this is the same as for Windows. You may need to make a exception rule for running Pronterface. The Pronterface app haven't a signed signature by Apple or Microsoft.
There is maybe a MacOS user around who can give you further help on this issue as I don't use MacOS.
Best regards,
DD

im-ellipsism, 2023-09-22:
> You forgot to mention the version of Pronterface, Is this the actual Pronterface version?
> I guess this is the same as for Windows. You may need to make a exception rule for running Pronterface. The [...]

I tried the two Mac versions that appear,
printrun-2.0.1_macos-11_x64_py3.10.zip
printrun-2.0.1_macos-12_x64_py3.10.zip

I'll see how to make an exception rule on Mac just in case.

thanks for your contribution.

DivingDuck, 2023-09-22:
This is the actual version of Pronterface.

This is maybe helpful: [https://support.apple.com/en-us/HT202491 ](url)

Please give a feedback whether the solve your problem.

Best regards,
DD

im-ellipsism, 2023-09-22:
That is what I did. First I found that problem apps can be opened by pressing control and clicking open from the context menu. Then I found the same thing you showed me, where I can create exceptions to open programs with problems.

I am very thankfull for your help.

Saludos!!! :)

closed 2023-09-22 by im-ellipsism